brianrho/FPM #1: ESP Crash when downloading template

danbicks, 2017-06-11:
Brian,

Your library is excellent, I have 2 readers of which both should be FPM10 format.

Old style works perfect downloading template to array, I have adapted this and now store to spiffs file and retrieve as well, works really well. One problem encountered with getTemplate is that when wifi connection is present sometimes times out or incorrect bytes returned 384 bytes instead of 512. Turning of wifi connection returns 512 template bytes every time.

I ordered some new readers which I thought were the same but ended up slightly different. Plugged in works in a normal way for all operations testes apart from getTemplate which crashes ESP with stack error.

new reader: http://www.ebay.co.uk/itm/322197616177?_trksid=p2060353.m1438.l2649&ssPageName=STRK%3AMEBIDX%3AIT

Byte format seems to be slightly different I think, have you tested this type of reader?

brianrho, 2017-06-13:
Any reason why you think the protocol is different? As far as I can tell, the module is basically the same as the one I used. Try the new modules with an Uno; if everything works as expected, then this is an ESP related issue, most likely caused by the absence of yield() calls in the loop where readRaw() grabs the template.

danbicks, 2017-06-14:
Brian,

Great Idea, I will try this with an Uno and also try adding yield()
statements. With ESP the standard finger print reader works perfect so
their must be something slightly different between the 2 types I would have
thought.

The reason I thought it could be protocol related is the byte's that are
received using an old routine of mine differ between the 2 reader types I
will look in to this further and give you the output and old function if
this helps.

I was looking to implement instead of spiffs Google firebase storage of
finger templates have you looked at firebase?

Big thanks

Dans

brianrho, 2017-06-14:
No I haven't used Firebase nor do I have much experience with ESPs really. The debug option in the library isn't well implemented yet else that would've been the way to go so I can see the bytes coming from the new reader. But you say everything works for both reader types except for the process of reading in the templates (I assume you don't mean the getTemplate() call itself) which causes problems both with the new and old readers when used with an ESP; that's why I think the issue is ESP-related. I guess we'll know for sure after you've tested with an Uno and inserted the yield()s.

danbicks, 2017-06-15:
Hi Brian,

Big thanks for getting back so soon, yes the getTemplate() routine works fine with the older reader type as long as wifi has no connectivity on the ESP, with the new reader type the routine causes a stack dump crash. Could be lacking of yield statements however the data output using one of my routines looks different. I will do some further tests over this weekend and beginning of next week and report back the results of my tests.

Keep you posted, thanks again for all your efforts.

Dans

commit ac7eeec6 (2018-11-10): Fix #1 of R551 getFreeIndex
closed 2017-12-02 by brianrho