brianrho/FPM #2: Uploading image to R305

amarpreetsingha, 2017-09-07:
Hii brianrho. Can you help me in uploading an image file to R305 using arduino. As i dont want to enroll by scanning through FPS. Thier is function DownloadFP in adafruit lib that imports image file to module i just need help in that.

amarpreetsingha, 2017-09-07:
Correction : downloadFP function definition is in datasheet not in adafruit lib.

brianrho, 2017-09-09:
Currently busy with other stuff. Will add that when I can.

sand007man, 2018-11-02:
> Hii brianrho. Can you help me in uploading an image file to R305 using arduino. As i dont want to enroll by scanning through FPS. R305 datasheet contains DownImage command description so just needed help in implementing that functionality.

hey were you able to implement that functionality using Arduino?? even I am stuck in that part??

commit ef05840d (2018-11-11): Fix #2 of R551 getFreeIndex
closed 2018-11-11 by brianrho