brianrho/FPM #3: I can not save over 164 fingerprints

juanlu46, 2017-11-28:
I have in poseseion a fingerprint sensor FPM10A, like this link (). In the description it says that it has capacity for 1000 fingerprints. The bookstore tells me that it has capacity for 300 prints.

However, when I try to store some fingerprint with an ID greater than 163, it tells me that it has been stored correctly, but then in the sketch "fingerprint" does not recognize the fingerprint (if it does when the ID is lower). When I pass the ID 300 it gives me an error of "FINGERPRINT_BADLOCATION" and it tells me that it can not save the fingerprint.

I use an Arduino DUE and the last commit of your library.

Thanks and regards!

brianrho, 2017-11-28:
What capacity is reported by the module itself, when you run an example
from the library?

juanlu46, 2017-11-28:
When I execute the example, the library tells me that the capacity is 300, but when I save the prints in position 255 for example, it tells me that it is correct and has been saved, but then it does not recognize it.

It is correct that the capacity is 300 ?, to claim the seller. Although I have also read that the capacity varies according to the level of security, I do not know if that is the case.

the link of the product in question, that I forgot before: https://es.aliexpress.com/item/FPM10A-Fingerprint-Reader-Sensor-Module-Optical-fingerprint-Fingerprint-Module-For-Arduino-Locks-Newest /32815608769.html?spm=a2g0s.9042311.0.0.WJ0aE5

brianrho, 2017-11-28:
> Reader-Sensor-Module-Optical-fingerprint-Fingerprint-
> Module-For-Arduino-Locks-Newest /32815608769.html?spm=a2g0s.

It seems the capacity reported by the module really isnt reliable. I
remember I was able to store up to 1649 prints in the first FPM10 I used,
despite the advertised 1000. If you can't save beyond 164, then there isn't
much that can be done. You can try changing the security level to see if it
helps though. I'll spend some time later to review how the library obtains
the capacity, just on the off-chance that it reports the wrong value.
You can also try the demo software (
http://www.rhydolabz.com/documents/widgets/R305_App_V2.0.zip) and see what
capacity it reports.

juanlu46, 2017-11-28:
I have tried the program that you have passed me, and uploading an example for communication with the program. The capacity shown is 300.

I have tried changing the level of security and the size of the packages but without being able to change the capacity. I have registered my fingerprint with ID 299 and in the quick search it has recognized me with this ID.

Even so, your library lets me save it with ID 299, but then it is not able to find that ID.

Thanks for your help!

brianrho, 2017-12-18:
I've made a number of changes to the library since then. Perhaps you can try it out again when you have the time?

juanlu46, 2017-12-19:
Ok, As soon as I have some time I check it again. Thanks for your time :)

closed 2018-11-10 by brianrho