brianrho/FPM #4: Change the LED Color or turn it off

Omega236, 2017-12-18:
Hello,
i see pictures where the LED is Red, my FPM is green.

is it possible to change the color? so if it read a finger it switch to red or turn off the LED?

brianrho, 2017-12-18:
Your module is a ZFM60? There are 6 wires in total coming out of the
module, right?

Omega236, 2017-12-18:
oh sorry, it is a FPM10A from Aliexpress.

But yes it has 6 wires

brianrho, 2017-12-18:
Okay I suspect it has the same interface as a ZFM60 anyways. So the LED
colour is fixed in all modules actually, at least as far as I know. Theres
a way to turn on/off the LED though. But the LED must be on for
enrollment/matching since its what illuminates the finger to be scanned
(hence the name, optical scanner). So if you turn off the LED with a
command, you must ensure you have some other way to know when a finger is
placed or about to be placed (a keypad or RFID card input, for instance),
so that you can then turn on the LED to begin scanning else fingers wont be
recognized or matched correctly.

Omega236, 2017-12-18:
Ok this is what i want. How do i turn off/on the led?

brianrho, 2017-12-18:
I've added 3 new methods: led_on(), led_off() and getImageNL(). The last
method does the same job as the old getImage() except this one does not
control the LED at all; it's left to you to turn on/off the LED as needed.
You can try them out with the enrollNL example, which I haven't tested yet.

Omega236, 2017-12-19:
Unfortunately, led_off () does not work. The LED remains active.

On the Board is DY50_MAIN_V3 and it has 6 red wires like in this PDF
https://cdn-learn.adafruit.com/downloads/pdf/adafruit-optical-fingerprint-sensor.pdf

brianrho, 2017-12-20:
Then it means it doesn't work for your module. I've only tested it successfully on a ZFM60 (actually found the commands in its datasheet). I tried it on my second module which is an FPM10A and neither led_on(), led_off() nor getImageNL() worked. Recognized commands depend on the device firmware so if you want to control the LED, it's best you buy a ZFM60.

closed 2018-01-16 by brianrho

edautz, 2019-02-25:
Great library. Good job.

Today I tested a FPM10A module marked with DY50 2V1 on the circuit board. It has also two extra touch pins called T-out and T-3,3V. The LED control functions worked out of the box using the led commands. The LED Control and touch control are easely integrated in the example code and usefull to trigger the sensor readout. Did a test with the enroll function and the search database function and encounterred no issues with Serial monitor.

I used an NodeMcu 1.0 ESP8266 with board version 2.4.2 and IDE 1.6.13

I will do some more testing with the other functions mentioned in the example code. After that I going to develop by using WiFi, ESPWebserver and websockets, an webinterface to create a browser based menu to use this sensor as a part of my alarm subsystem of my domotic system.

edautz, 2019-02-27:
Tested other functions like delete and empty database, workes fine, sometimes a time-out, could be caused by softserial library.

Did a quick test on a ESP32 devkit-V1. First test with enroll, match, delete database, count fingerprints etc, looked good. Touch and ledcontrol working fine.

brianrho, 2019-02-27:
Great. Maybe you can also test the image download to RAM on the ESP32, should have the needed 37k I think.  How often do you get the timeouts and for which commands? Do you have debugging enabled when they happen?

edautz, 2019-02-28:
Ok. What do you mean with “test the image download to RAM”, like the finger.loadModel(fid, 2); function like in the match_pair.ino example?

The tests on the ESP8266 was only a test to check if this library works on my purchased fingerprint sensor. I didn’t turn debugging on, because I suspect the softserial library cannot chatch up all the time when using the highest baudrate. I had to use it because of the lack of two hardware serials on the Esp8266 when testing your examples modified with the touch en the led functions.

On the ESP32 I used two hardware serials. I Will do some more testing before moving on with my project.

brianrho, 2019-02-28:
Yes, high baud rates will also cause timeouts, has to do with the nature of timing with SoftwareSerial, compared to Serial. 
Regarding the image download, I was referring to the image_to_pc example. It can be modified to write the entire image to a buffer in RAM rather than to the Serial port.

edautz, 2019-02-28:
OK. I see what you mean. 

I think from the example the function: void stream_image(void) has te be altered or replaced to do the image download to RAM.

Frankly I don't have any experience to program such an task.  If you provide a code example, I can test it out in my ESP32 setup.

brianrho, 2019-02-28:
Okay, try this out when you can: https://pastebin.com/YXDgEhdV. If it works, it should take about 6 secs to read in the entire 37k image into the buffer (at 57600 baud) and then print the total size read

edautz, 2019-02-28:
Very good. I will try this code soon. I noticed on the ESP32 also random time-outs. Mostly when I am using the sensor to verify a print.
After that I am not able to communicatie with the sensor anymore and have to reset it.
I am going to enable debug and try to catch an error. But it is hard to reproduce. The sensor operates on the default baudrate of 57600 bps 8N1. Does it matter to lower it to 38400 bps or even lower?

brianrho, 2019-02-28:
My FPM10 needs 5V, is yours labelled 3.3V? Debugging will probably hide the issue, the way its written at the moment. Send a screenshot of what you see when the timeouts occur, probably best to create a new issue

edautz, 2019-03-01:
Here are results of the image download test:

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10088
load:0x40080400,len:6380
entry 0x400806a4
SEND IMAGE TO PC test
Found fingerprint sensor!
Capacity: 150
Packet length: 128
Packet length set to 128 bytes
Waiting for a finger...
Image taken
Starting image stream...
Timestamp: 4719

Timestamp: 12444
36864 bytes read.
Image stream complete.

Took about 7,7 seconds

If I can do some more testing let me know. Today no timeouts .........

brianrho, 2019-03-04:
Okay then, good to know the image download to a buffer works. If you get any timeouts in future, create a new issue and provide screenshots of the serial output.

brianrho, 2019-03-04:
Also make sure that your sensor really works with 3.3V and not 5V.

edautz, 2019-03-04:
It is a 3,3V for shure. It is printed on the PCB. One of the connection is labeled 3V3 and the touch power connection is labeled T-3V3.

On this moment I am embedding this sensor in my code, so I can use WiFi and a webinterface to enroll and delete the prints. I use a remote debug library, so I don’t need the second serialport for debugging and can do a WiFi based debug at any place.

Together with a OLED display and a 4 digit touchpad, I wil integrate this in my alarm system.

I have to do a lot of more testing to get a stable system.

I noticed that the example codes contains yield statement, but these don’t prevents watchdog time-outs in a ESP32. I replaced them with delay(1), that proves for now preventing the watchdog time-outs.

When the time-out errors persist I going to add a mosfet to my setup, so I can switch the power for the fingerprint sensor to hardware reset is.

Do you know a way to software reset the fingerprint sensor? That would be a simpeler solution.

brianrho, 2019-03-04:
Wait, I thought the timeouts you spoke of were for the FPM commands. You mean, the ESP32 itself restarts from a watchdog reset?

edautz, 2019-03-04:
Both situations have occurred. When I used only code based on Serial communication, some time-outs occurs in the Serial communication to the fingerprint sensor.

After bringing the WiFi code and a webserver live, watchdog errors start to occurs, causing by while loops and yield combinations. Solved this to replace the yields by a delay(1).

brianrho, 2019-03-04:
The ESP32 generally isnt vulnerable to WDT resets like the ESP8266, a separate core is used for WiFi tasks, it doesnt need those yield() statements, so I was surprised when you said it timed out because of the WDT.

But if it works now, thats fine then

kennethallag, 2019-03-25:
hello, how to use the ledOn(), ledOff()... I'm using the ZFM040_V1.6 fingerprint sensor.
I want to know how to control the led.  I'm only a newbie. thanks

brianrho, 2019-03-25:
Try the `enroll_with_led_control` example. If it raises errors and doesn't work for you, then your sensor doesn't support those commands. So far, the commands have only worked with the ZFM60 and DY50 sensors though.

kennethallag, 2019-03-28:
Where can I find these "enroll_with_led_control" example?? can you give me the link? Thanks

brianrho, 2019-03-28:
It's in the examples folder.

kennethallag, 2019-03-29:
Got it. Thanks.