brianrho/FPM #7: Touch

beryindo, 2018-09-11:
How to Use Touch PIN?

brianrho, 2018-09-11:
What touch pin?

closed 2018-10-02 by brianrho

edautz, 2019-02-26:
> How to Use Touch PIN?

https://github.com/brianrho/FPM/issues/4#issuecomment-467225878

devegied, 2020-06-12:
FPM10A Fingerprint Reader Sensor Module from AliExpress have two more pins: T-OUT and T-3V3. These are for integrated touch sensor.
I done some RE on part of sensor: [image]
The 233DH TonTouch IC is used for touch sensor and with this configuration (TOG - low, AHLB - high) on output T-OUT we should get HIGH when there is nothing on fingerprint sensor and HIGH when finger is on the sensor.