brianrho/FPM #8: R551 support

xsrf, 2018-09-21:
Hey,
I recently ordered a R551 module ( https://www.aliexpress.com/-/32879785395.html ) and it also works fine, as far as I can tell ( just tested enroll / read fingerprint so far ).

I found that sending command 0x33 or 0x60 turn the module "off" until it detects a finger. This also turns the LED and UART off. I cannot find this mentioned in any datasheet.

Also found that sending 0x47, 0x50, 0x51, 0x52, 0x53 or 0x68 returns something - but I don't know what it does.

The module has dual color LEDs (red/green) but I haven't found a way to control them... It's always green, or off in Stand-By.

```
---> 0xEF 0x1 0xFF 0xFF 0xFF 0xFF 0x1 0x0 0x3 0x33 0x0 0x37
<---  0xEF 0x1 0xFF 0xFF 0xFF 0xFF 0x7 0x0 0x3 0x0
Sent command 0x33 got 0x0 // StandBy?!

---> 0xEF 0x1 0xFF 0xFF 0xFF 0xFF 0x1 0x0 0x3 0x47 0x0 0x4B
<---  0xEF 0x1 0xFF 0xFF 0xFF 0xFF 0x7 0x0 0x13 0x0 0x50 0x50 0x4D 0x50 0x44 0x47 0x36 0x32 0x53 0x32 0x30 0x30 0x56 0x31 0x30 0x30
Sent command 0x47 got 0x0

---> 0xEF 0x1 0xFF 0xFF 0xFF 0xFF 0x1 0x0 0x3 0x50 0x0 0x54
<---  0xEF 0x1 0xFF 0xFF 0xFF 0xFF 0x7 0x0 0x3 0x0
Sent command 0x50 got 0x0

---> 0xEF 0x1 0xFF 0xFF 0xFF 0xFF 0x1 0x0 0x3 0x51 0x0 0x55
<---  0xEF 0x1 0xFF 0xFF 0xFF 0xFF 0x7 0x0 0x3 0x0
Sent command 0x51 got 0x0

---> 0xEF 0x1 0xFF 0xFF 0xFF 0xFF 0x1 0x0 0x3 0x52 0x0 0x56
<---  0xEF 0x1 0xFF 0xFF 0xFF 0xFF 0x7 0x0 0x3 0x2
Sent command 0x52 got 0x2

---> 0xEF 0x1 0xFF 0xFF 0xFF 0xFF 0x1 0x0 0x3 0x53 0x0 0x57
<---  0xEF 0x1 0xFF 0xFF 0xFF 0xFF 0x7 0x0 0x3 0x55
Sent command 0x53 got 0x55

---> 0xEF 0x1 0xFF 0xFF 0xFF 0xFF 0x1 0x0 0x3 0x60 0x0 0x64
<---  0xEF 0x1 0xFF 0xFF 0xFF 0xFF 0x7 0x0 0x3 0x0
Sent command 0x60 got 0x0 // StandBy?!

---> 0xEF 0x1 0xFF 0xFF 0xFF 0xFF 0x1 0x0 0x3 0x68 0x0 0x6C
<---  0xEF 0x1 0xFF 0xFF 0xFF 0xFF 0x7 0x0 0x5 0x0 0x0 0x2
Sent command 0x68 got 0x0
```

brianrho, 2018-09-22:
They're probably just hidden commands, occupying the reserved range. But I remember 0x50 and 0x51 are for turning on/off the LED for ZFM60 modules, perhaps that's also the case here. But there are no issues with the library, right?

xsrf, 2018-09-22:
0x50 and 0x51 don't do anything, the green led stays lit... Rest of the library works so far.
Maybe 0x33 is a command worth adding to the library?

xsrf, 2018-09-22:
I cannot find any datasheet for the R551 module or the µC PM6542A which it uses. I've contacted the seller as well as GROW directly but haven't got a response yet. I primary look for a way to control the red LEDs ...
I measured ~10µA with a multimeter, but it was permanently varying between 5µA and 15µA. I guess it's pulsed with few Hz. I'll measure again using an oszi. 
So far I've only used the module for few hours and I've never used a fingerprint sensor before, so I'm not sure if I know what I'm doing ;)

xsrf, 2018-09-23:
Okay, it pulses every ~250ms ... Smoothed out using a 100µF cap gives pretty much 10µA power consumption.

[image]
[image]
[image]

Regarding current, 0x33 and 0x60 are also identical

xsrf, 2018-09-23:
Okay, now I also found some problems with the library. `get_free_id()` always returns `0`, so every fingerprint is enrolled to `ID #0` using your example.
Calling `getFingerprintEnroll(1)` in your example however works fine to enroll a fingerprint into `ID #1` and both can be used afterwards.

Your "show_fingerprint_template" also has issues. I've enrolled `#0` and `#1` and set it to move `#1` to `#3`:
```
fingertest
Found fingerprint sensor!
Capacity: 500
Packet length: 128
Send any character to continue...
template 1 loaded
Error receiving packet
Template 1 deleted
Unknown error
```
After that, `#1` was deleted and not recognized anymore.

brianrho, 2018-09-26:
Regarding 0x33 and 0x60, it's unlikely that there are 2 commands which do the exact same thing, would be nice if you could figure out the difference between them. Still, I can add the commands as a sort of "light sleep" option. The library is due for a lot of re-structuring anyways.

Regarding the get_free_id(), when you say it returns 0, you're not referring to the literal return value of the function, right? I assume you know the function itself returns false if it couldn't find any slot and that the actual free id, if one is found, is stored in the passed `int16_t` argument. In that case, if the function really is storing 0 in the ID argument, did it print "Free slot at ID 0"? I'd like to know at what point the function failed. The english datasheet itself says nothing about a command to get the free slots but this by itself doesn't mean anything since that was also the case for the other modules; I had to use the original google-translated chinese datasheets. I can't find any original for the R551 though.

"Error receiving packet" means the initial load and upload commands succeeded but the transfer of the template failed at some point, maybe because of an incorrect packet length or even a UART timeout. You can try printing everything thats received beyond the "template 1 loaded" so we can see what arrived, if anything.

xsrf, 2018-10-11:
Hey,
I finally got a response from GROW... the R551 module was custom made for a customer of GROW who asked not to disclose information about the firmware to the public. This is why GROW only mentions the standard commands almost all their fingerprint modules have in the datasheet.
There won't be any official information or documentation about the 10µA stand-by command or the red LEDs :(

brianrho, 2018-10-11:
Shame. Will add them anyways when I can, so anybody can test their behaviour and maybe we can figure out the difference between them. You can try the recent library version and examples to see if they now work for the module, especially the templates example.

Snipeye, 2018-11-08:
I was just following up to see if there had been any progress made on this?  I'm about to buy one of these modules, and given the (terrible) documentation available, this is a goldmine for me.  I'll be happy to do some testing when I get it...

Snipeye, 2018-11-09:
I was also wondering if anything ever came of the red/green LEDs, as well.  Thanks for the response!

brianrho, 2018-11-09:
You're welcome. Since I don't have your module, there's not a lot I can do about the LEDs and I can only spend so much time on this. You can of course try out stuff and if you discover anything interesting, pls let me know.

xsrf, 2018-11-10:
Hey, sorry for being so quiet. I've downloaded your latest version today and now I can't get the sensor to read my fingerprint at all...

I used your examples as last time, enroll and then search_database:

```
Searching for a free slot to store the template...
Free slot at ID 0
Waiting for valid finger to enroll
...............Image taken
Image converted
Remove finger
Place same finger again
............................Image taken
Image converted
Prints matched!
ID 0
Unknown error
Send any character to enroll a finger...
Searching for a free slot to store the template...
Free slot at ID 0
Waiting for valid finger to enroll
.............................Image taken
Image converted
Remove finger
Place same finger again
.......................Image taken
Image converted
Prints matched!
ID 0
Stored!
Send any character to enroll a finger...
Searching for a free slot to store the template...
Free slot at ID 0
Waiting for valid finger to enroll
.......................Image taken
Image converted
Remove finger
Place same finger again
.......................Image taken
Image converted
Prints matched!
ID 0
Stored!
Send any character to enroll a finger...
```
```
MATCH test
Found fingerprint sensor!
Capacity: 500
Packet length: 128
Send any character to search for a print...
Waiting for valid finger
...................................Image taken
Image converted
Remove finger

Got wrong PID or length!
Send any character to search for a print...
Waiting for valid finger
............................Image taken
Image converted
Remove finger

Got wrong PID or length!
Send any character to search for a print...
```

enroll with debug:

```
[... 411 lines cut ...]
[+]Got header
[+]Address: 0xFFFFFFFF
[+]PID: 0x7
[+]Length: 1
0
[+]Read complete
Image converted
Remove finger

[+]Got header
[+]Address: 0xFFFFFFFF
[+]PID: 0x7
[+]Length: 1
2
[+]Read complete

[+]Got header
[+]Address: 0xFFFFFFFF
[+]PID: 0x1
[+]Length: 6
1B 1 0 0 1 F4
[+]Read complete
Got wrong PID or length!
Send any character to search for a print...
```

brianrho, 2018-11-10:
I think I know why the fingerprint search didn't work, basically your module seems to not support the "high-speed" search command 0x1B, so it just echoed the command and parameters it received. I've fixed that in the library, you just need to comment the `FPM_ENABLE_HISPEED_SEARCH` macro and it will use the regular search command 0x04. 
As for the failed enrollments, I'm not sure what the problem is, I haven't changed any of its commands, there are no errors in the process and enrollments work just fine on my end. Maybe you can fix the search error first, enroll at a different location and try again, if it fails, send the debug messages for 2 successive enrolls at least.

xsrf, 2018-11-10:
Thank you, with `FPM_ENABLE_HISPEED_SEARCH` commented out it works again. I can enroll a fingerprint and it can be found. I also can manually enroll a fingerprint to slot 1 and it can also be found.
`get_free_id()` however still returns slot 0 all the time.
You already have the debug log from `get_free_id()` in my post above, can I give you additional debug info?

brianrho, 2018-11-10:
Sure, in debug mode, try multiple enroll attempts calling get_free_id() each time but not using the ID it returns, instead set the location manually in the sketch. Also try using your previous version of the library ([here, I think](https://github.com/brianrho/FPM/tree/994febc664e190223f3d68335bd34e2060237e98)) to be sure get_free_id() worked properly then, if it did, then send those debug messages too.

xsrf, 2018-11-10:
Okay, I now called `empty_database();` at the beginning and then enrolled 4 fingers into ID 0,1,2,3.
I changed the debug output to be less verbose:

```
ENROLL test
[+]Length: 1; Data: 0
[+]Length: 17; Data: 0 0 0 0 0 1 F4 0 3 FF FF FF FF 0 2 0 6
[+]Length: 17; Data: 0 0 0 0 0 1 F4 0 3 FF FF FF FF 0 2 0 6
Found fingerprint sensor!
Capacity: 500
Packet length: 128
[+]Length: 1; Data: 0
Database empty!
Send any character to enroll a finger...
Searching for a free slot to store the template...
[+]Length: 33; Data: 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
Free slot at ID 0
Waiting for valid finger to enroll
[+]Length: 1; Data: 0
Image taken
[+]Length: 1; Data: 0
Image converted
Remove finger
[+]Length: 1; Data: 2
Place same finger again
[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 0
Image taken
[+]Length: 1; Data: 0
Image converted
[+]Length: 1; Data: 0
Prints matched!
ID 0
[+]Length: 1; Data: 0
Stored!
Send any character to enroll a finger...
Searching for a free slot to store the template...
[+]Length: 33; Data: 0 2 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
Free slot at ID 0
Waiting for valid finger to enroll
[+]Length: 1; Data: 0
Image taken
[+]Length: 1; Data: 0
Image converted
Remove finger
[+]Length: 1; Data: 2
Place same finger again
[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 0
Image taken
[+]Length: 1; Data: 0
Image converted
[+]Length: 1; Data: 0
Prints matched!
ID 1
[+]Length: 1; Data: 0
Stored!
Send any character to enroll a finger...
Searching for a free slot to store the template...
[+]Length: 33; Data: 0 6 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
Free slot at ID 0
Waiting for valid finger to enroll
[+]Length: 1; Data: 0
Image taken
[+]Length: 1; Data: 0
Image converted
Remove finger
[+]Length: 1; Data: 2
Place same finger again
[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 0
Image taken
[+]Length: 1; Data: 0
Image converted
[+]Length: 1; Data: 0
Prints matched!
ID 2
[+]Length: 1; Data: 0
Stored!
Send any character to enroll a finger...
Searching for a free slot to store the template...
[+]Length: 33; Data: 0 E 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
Free slot at ID 0
Waiting for valid finger to enroll
[+]Length: 1; Data: 0
Image taken
[+]Length: 1; Data: 0
Image converted
Remove finger
[+]Length: 1; Data: 0
[+]Length: 1; Data: 2
Place same finger again
[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 2
.[+]Length: 1; Data: 0
Image taken
[+]Length: 1; Data: 0
Image converted
[+]Length: 1; Data: 0
Prints matched!
ID 3
[+]Length: 1; Data: 0
Stored!
Send any character to enroll a finger...
```
The 2nd byte of the response changes.

`get_free_id()` didn't work with the earlier versions (it never worked for me so far).

brianrho, 2018-11-10:
I see a pattern and the problem I think. The LSB remains free but the table of stored IDs does get updated, just off by a bit. Don't wipe the database, try enrolling with IDs 6, 7, 8 so I can confirm the nature of the problem. Send those debug messages too

xsrf, 2018-11-10:
Guess that's enough? :)
```
Send any character to enroll a finger...
Searching for a free slot to store the template...
[+]Length: 33; Data: 0 1E 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
...
Prints matched!
ID 6
Stored!
Send any character to enroll a finger...
Searching for a free slot to store the template...
[+]Length: 33; Data: 0 9E 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
...
Prints matched!
ID 7
Stored!
Send any character to enroll a finger...
Searching for a free slot to store the template...
[+]Length: 33; Data: 0 9E 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
...
Prints matched!
ID 8
Stored!
```

brianrho, 2018-11-10:
Yeah, it looks like an off-by-one error. For your module, ID 0 corresponds to bit 1 in the index table and it counts as usual from there. Bit 0 is always cleared so get_free_id() returns immediately since that's the first bit it inspects. Will fix and push in a few mins.

brianrho, 2018-11-10:
Can you try enrolling at IDs 254, 255, 256, 257?

brianrho, 2018-11-10:
I've made some changes, should work for IDs up to the first page's boundary at 254. Whether it will work beyond that depends on the results of the enrollments I requested earlier

xsrf, 2018-11-10:
Okay, I'm running out of fingers :D 
For 254 and 255 I used a finger I used before... and got an error back. 256 and 257 stored fine

```
Searching for a free slot to store the template...
[+]Length: 33; Data: 0 9E 3 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
...
Prints matched!
ID 254
[+]Length: 1; Data: 24
Unknown error
Searching for a free slot to store the template...
[+]Length: 33; Data: 0 9E 3 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
...
Prints matched!
ID 255
[+]Length: 1; Data: 24
Unknown error
Searching for a free slot to store the template...
[+]Length: 33; Data: 0 9E 3 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
...
Prints matched!
ID 256
[+]Length: 1; Data: 0
Stored!
Searching for a free slot to store the template...
[+]Length: 33; Data: 0 9E 3 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
...
Prints matched!
ID 257
[+]Length: 1; Data: 0
Stored!
Searching for a free slot to store the template...
[+]Length: 33; Data: 0 9E 3 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
```
guess 0x24 means the fingerprint is already saved...

now, having only one finger unused, I tried 254 again

```
 Searching for a free slot to store the template...
[+]Length: 33; Data: 0 9E 3 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
...
Prints matched!
ID 254
Stored!
Send any character to enroll a finger...
[+]Length: 33; Data: 0 9E 3 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 80
```
now slots 0 1 2 3 6 7 8 254 256 257 are used

brianrho, 2018-11-10:
Your module is getting stranger, other modules so far don't care if you enroll the same finger multiple times. I imagine it was a feature the client requested. You can wipe the database if you need to. Since get_free_id() isn't really being used now, replace its current body with just this:

```
finger.getFreeIndex(0, fid);
finger.getFreeIndex(1, fid);
return true;
```

Then enroll 254 - 257 again

xsrf, 2018-11-11:
I'm surprised other modules don't complain. How do they match a fingerprint if it is stored in more than one slots?

Anyways, here is the debug log:

```
Database empty!
[+]Length: 33; Data: 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
[+]Length: 33; Data: 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 EF
Prints matched!
ID 254
Stored!
[+]Length: 33; Data: 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 80
[+]Length: 33; Data: 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 EF
Prints matched!
ID 255
Stored!
[+]Length: 33; Data: 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 80
[+]Length: 33; Data: 0 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 EF
Prints matched!
ID 256
Stored!
[+]Length: 33; Data: 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 80
[+]Length: 33; Data: 0 3 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 EF
Prints matched!
ID 257
Stored!
[+]Length: 33; Data: 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 80
[+]Length: 33; Data: 0 7 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 EF
```

brianrho, 2018-11-11:
I think they just return the first location where the fingerprint is found, it's up to the user to scan before enrollment and ensure a print doesn't already exist. But you can of course use your module's return code to check for that, you just need an additional case for the 0x24 return value in the switch statement.

I've made the changes, you can now test get_free_id() as usual in enrollment to see if it works as expected, though unless you can enroll up to 256 fingers, all get_free_id() calls will return free IDs in page 0 (ID 0 - 254). To test the next page 1 (ID 255 - 503, it looks like), you can enroll (and delete alternately) at addresses >= 255, then explicitly call finger.getFreeIndex() on page 1 to see if it returns the correct free IDs after each enrollment.

xsrf, 2018-11-14:
@brianrho I'm sorry, I just had 10min to test it... but yes, the enroll example now works fine :)

closed 2018-11-15 by brianrho

henfri, 2019-06-29:
Hello,

So ist this module now fully supported, with the exception of the LED?
Can you recommend it?

Regards
Hendrik

brianrho, 2019-06-30:
@henfri The situation's the same as stated in the readme. Advanced functionality like image and template downloads still don't work. But if you all want to do is perform enrollments and matching, then you can go ahead. Anything else and you should just get an FPM10 or R307.

adarshparavoor, 2019-07-30:
can any one send me the arduino working  code for this sensor.

ankit993, 2020-01-15:
i am keeping search_database() at main loop, so i can search my fingerprint and send it to server, but every 8 minutes, it disconnects mqtt and not connect to server, then i have to hard reset mcu,

brianrho, 2020-01-15:
@ankit993 Create a new issue and provide all the details, including your code and sensor info.