brianrho/FPM #9: show_fingerprint_template

milindrc, 2018-09-28:
Hey, your sketch can solve all my problems, But unfortunately it's not working although all other sketches are working but I just couldn't get this sketch to work. I am getting the following error

template 1 loaded
Error receiving packet
Unknown error

digging in your library i found that this snippet is causing the function to close before end packet.
`        if (bufLen == NULL || *bufLen < packetLen)        // if a buffer was provided, ensure theres enough space
            {
			return false;
        }else
`

I tried to fix it but for some reason my arduino crashes.

Please help me out.

Thanks in advance.

brianrho, 2018-09-28:
Are you running the exact example? Do you have a print already enrolled at ID 1?

milindrc, 2018-09-29:
yes a double checked everything, as i said all the other examples are working, but just this one.

brianrho, 2018-09-29:
Add `Serial.println(*buflen); Serial.println(packetLen);` inside the IF statement, before `return False`, and let me see the results.

milindrc, 2018-09-30:
https://www.amazon.in/REES52-Internet-Development-Wireless-Compatible/dp/B071ZJ8Z5L/ref=sr_1_7?s=industrial&ie=UTF8&qid=1538326709&sr=1-7&keywords=nodemcu

milindrc, 2018-09-30:
packet length was 128

brianrho, 2018-09-30:
No, add the lines and let me see the actual results in the serial monitor.

brianrho, 2018-09-30:
Also provide a link to your fingerprint module

milindrc, 2018-10-01:
Sorry sent you the wrong link, this is the one, 

Robocraze 5 Optical Fingerprint Reader Sensor Module for Mega2560 UNO R3 Raspberry Pi 3 RC-A-400 https://www.amazon.in/dp/B07DL2DXJX/ref=cm_sw_r_cp_apa_i_UkySBbTS58S6T

brianrho, 2018-10-01:
It seems you're passing 0 for buffer size. I need to see the *exact* sketch you're executing

milindrc, 2018-10-01:
Just using your example

#include <SoftwareSerial.h>
#include <FPM.h>

#define BUFF_SZ 512

#define TEMPLATE_TO_MOVE            0
#define NEW_TEMPLATE_LOCATION       4

// Download a template, delete it, print it, and upload it to a different flash location

// pin #2 is IN from sensor (GREEN wire)
// pin #3 is OUT from arduino  (WHITE/YELLOW wire)
SoftwareSerial mySerial(2, 3);

FPM finger;

void setup()  
{
  Serial.begin(9600);
  Serial.println("fingertest");
  mySerial.begin(57600);
  
  if (finger.begin(&mySerial)) {
    Serial.println("Found fingerprint sensor!");
    Serial.print("Capacity: "); Serial.println(finger.capacity);
    Serial.print("Packet length: "); Serial.println(finger.packetLen);
  } else {
    Serial.println("Did not find fingerprint sensor :(");
    while (1) yield();
  }
  Serial.println("Send any character to continue...");
  
  while (Serial.available() == 0) yield();
  
  getTemplate(TEMPLATE_TO_MOVE);  // download template at #4 (if it exists) to the buffer; first enroll a finger at that location
//  deleteTemplate(TEMPLATE_TO_MOVE); // delete template at #4
  sendTemplate(NEW_TEMPLATE_LOCATION);   // upload template in buffer to #8; run fingerprint match example to verify that the new template is now at #8
}

void loop()
{

}

uint8_t buff[BUFF_SZ];

void getTemplate(uint16_t id)
{
  uint8_t p = finger.loadModel(id);
  switch (p) {
    case FINGERPRINT_OK:
      Serial.print("template "); Serial.print(id); Serial.println(" loaded");
      break;
    case FINGERPRINT_PACKETRECIEVEERR:
      Serial.println("Communication error");
      return;
    default:
      Serial.print("Unknown error "); Serial.println(p);
      return;
  }

  // OK success!

  p = finger.getModel();
  switch (p) {
    case FINGERPRINT_OK:
      break;
   default:
      Serial.print("Unknown error "); Serial.println(p);
      return;
  }
  
  bool last;
  int count = 0;
  uint16_t buflen = 512;
  uint16_t pos = 0;
  
  while (true){
    bool ret = finger.readRaw(buff + pos, ARRAY_TYPE, &last, &buflen);
    if (ret){
      count++;
      pos += buflen;
      buflen = BUFF_SZ - pos;
      if (last)
        break;
    }
    else {
      Serial.println("Error receiving packet");
      return;
    }
    yield();
  }

  Serial.println("---------------------------------------------");
  for (int i = 0; i < 32; i++){
    for (int j = 0; j < 16; j++){
      Serial.print(buff[i*16 + j], HEX);
      Serial.print(" ");
    }
    Serial.println();
    yield();
  }
  Serial.println("--------------------------------------------");

  Serial.print(count * finger.packetLen); Serial.println(" bytes read");
}

void deleteTemplate(uint16_t id){
  int p = finger.deleteModel(id, 1);
  switch (p){
    case FINGERPRINT_OK:
      Serial.print("Template "); Serial.print(id); Serial.println(" deleted");
      break;
    case FINGERPRINT_PACKETRECIEVEERR:
      Serial.println("Comms error");
      break;
    case FINGERPRINT_BADLOCATION:
      Serial.println("Could not delete from that location");
      break;
    case FINGERPRINT_FLASHERR:
      Serial.println("Error writing to flash");
      break;
    default:
      Serial.println("Unknown error");
  }
  return;
}

void sendTemplate(uint16_t id){
  int p = finger.uploadModel();
  switch (p) {
    case FINGERPRINT_OK:
      Serial.println("Starting template upload");
      break;
    case FINGERPRINT_PACKETRECIEVEERR:
      Serial.println("Comms error");
      return;
    case FINGERPRINT_PACKETRESPONSEFAIL:
      Serial.println("Did not receive packet");
      return;
    default:
      Serial.println("Unknown error");
      return;
  }

  yield();
  finger.writeRaw(buff, BUFF_SZ);

  p = finger.storeModel(id);
  switch (p) {
    case FINGERPRINT_OK:
      Serial.print("Template stored at ID "); Serial.println(id);
      break;
    case FINGERPRINT_PACKETRECIEVEERR:
      Serial.println("Comms error");
      break;
    case FINGERPRINT_BADLOCATION:
      Serial.println("Could not store in that location");
      break;
    case FINGERPRINT_FLASHERR:
      Serial.println("Error writing to flash");
      break;
    default:
      Serial.println("Unknown error");
  }
  return;
}

brianrho, 2018-10-01:
Hmm. Add a Serial.

milindrc, 2018-10-01:
I didn't get you, add what and where?

sand007man, 2018-10-02:
hi , Even i am facing the same issue after adding the Serial print i got this output

⸮txiR,h⸮⸮@>h⸮:⸮⸮fingertest
Found fingerprint sensor!
Capacity: 300
Packet length: 128
Send any character to continue...
template 4 loaded
Count: 0
Count: 1
Count: 2
Count: 3
Error receiving packet

brianrho, 2018-10-02:
@sand007man Okay change:

```
#define BUFF_SZ 512
```

to:

```
#define BUFF_SZ 2048
```

and:

```
uint16_t buflen = 512;
```
to:

```
uint16_t buflen = BUFF_SZ;
```

in the sketch and let me see what it prints.

sand007man, 2018-10-02:
hi i was not able to find uint16_t buflen variable in your code, but i did made the change in BUFF_SZ = 2048 and got this result

3 3 58 2C 0 1 20 1 82 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
8 0 0 0 81 0 0 0 0 0 C FC CC FF CF FF 
FF FF FF FF EE AA AA AA AA AA AA 66 65 55 55 55 
55 55 15 45 14 44 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
29 84 0 7E 53 86 5A 7E 5E A 46 3E 43 8A 98 DE 
55 91 46 DE 5F 26 5F 1E 1E 29 C 1E 1D 2D 22 9E 
66 37 CA FE 6C 41 8A 7E 74 43 7 BE 1F 86 54 BF 
2E A 54 BF 5F 1B C8 5F 4E 1D 8 3F 6E AB 88 5F 
15 B1 89 7F 2E B9 90 3F 39 BA F 1F 61 3C 8C 1F 
61 AC CB 7C 28 35 90 1C 38 1D 97 9D 3E 22 49 FD 
3B 2E 8D BD 20 B2 63 3D 2E 1C 80 1A 29 A3 8D 5A 
28 AC A3 BA 68 2E 49 5A 41 9F 88 BB 31 23 4B 7B 
32 B3 23 F8 1A B8 21 18 30 A0 5 F9 34 AF 4E 19 
1A 35 A0 39 14 BD DE F7 20 40 66 17 2C 2F 90 F4 
2E B1 23 D4 1D 3C 24 B5 1B 3E 61 D5 16 40 DD 93 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
3 3 56 1B 0 1 20 1 7E 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
11 0 0 0 71 0 0 0 0 33 FF FF FF FF FF FF 
FF FF BA AA AA AA A6 65 65 59 55 55 55 51 55 15 
10 44 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
--------------------------------------------
768 bytes read.
Template 4 deleted
Enter the template's new ID...

brianrho, 2018-10-02:
Ah, I see the problem now I think. I assumed the template size on all modules is 512 bytes, based on the datasheets I've read but it seems some modules have a template size of 768 bytes. Will adjust the example accordingly in a minute. 

@milindrc you should download and try the latest version, should hopefully work for you now.

brianrho, 2018-10-02:
The changes have been made to the example, you can test them (and the others again) to see if they work for you.

sand007man, 2018-10-02:
hi, Thank you for all the help and the changes you made works great :).. thank you so much :)

milindrc, 2018-10-02:
Now it workd flawlessly!!!!
I can't thank you enough.
You are a lifesaver <3

brianrho, 2018-10-02:
You're welcome ;-)

closed 2018-10-02 by brianrho

sand007man, 2018-10-03:
Hi i know this is out of topic but need a bit of help if possible to store this template value in mysql database. So my first step should be to store the buffer value in some other variable and then from there i have to update it to mysql database using inser query...!!but how to store 768 byte of data!!

sand007man, 2018-10-03:
Ok will try that but i have to end up using external eeprom or sd card to do that

brianrho, 2018-10-03:
Isn't your database external on some server?

sand007man, 2018-10-03:
Locally(mysql 5.7 server) and my node_mcu will be connected to the same network

brianrho, 2018-10-03:
So stream the contents of the template array to your PC server over the network and have it write the 768 bytes to a file, to be saved in your database. No need for an sd card or anything.

sand007man, 2018-10-03:
Ok "Stream"....that concept something new to me, when it comes to node_mcu...how is it possible?? any guidance...my node_mcu will be connected to the network via router ...i did google about file transfer over the network most of it suggested a php script i have to write.I am able to insert all other sensor value i the database just 768 byte of data i dont kn how to do this !!

brianrho, 2018-10-03:
You're not transferring a file, you're transferring an array's contents, mere bytes. Check your Node MCU `WiFiclient` examples to see how data is sent to a server, probably something like `client.write(buffer, buffer_sz);`, where `buffer` is the name of the template array and `buffer_sz` is the number of bytes/template. It's up to your server to then receive the bytes and then write them to a file.

sand007man, 2018-10-05:
Hi thanks for the all the input but i figured it out how to post this template directly in mysql database ..all i had to do is to conevert the buffer value into char array and upload it to mysql database :)

sand007man, 2018-10-06:
Hi sorry to disturb you again but need a small help in template comparison, I have two template stored in database and what to compare it ...do you have any idea about fingerprint confidence match that happens in adafruit library ...can the same logic be implemented for the template on the server side ...i kn i am asking for too much help !!

sand007man, 2018-10-30:
hi, sorry i got stuck in template format...as in there is 2 fingerprint template format ...ANSI 381...and ISO 19794-2....which is the format generated by this code....or is it hardware specific????

brianrho, 2018-10-30:
It's not a detail that's known, manufacturer secret. So basically you cant compare templates on your PC since you dont know the rules for comparing them. You're of course free to experiment and try the different formats and see if they hopefully used one of them. Or you can perform a 1:1 match by having your server load the user's enrolled print into the module buffer and having the module compare that against the presented finger. This method assumes that you have some other means of identifying the user, maybe some RFID card that's presented by the user and used to call up their enrolled print. In any case, for now, the module performs all comparisons between templates.

sand007man, 2018-10-31:
Ok I have stored the template in the database ..i can get that template using sql query and stored in my esp8266 .. so to compare that template with the current template generated i have to upload the template from server to the fingerprint module....so i did little research and there is something upchar and downchar in fingerprint module to upload the other template !! ....and for finger.match_pair() what are the arguments to be passed??

brianrho, 2018-11-08:
I've added an example for match_pair(), see if it works for you.

thiagoanselmo, 2019-01-03:
I'm already using soureceAFIS, however the time to extract the image and send via network to a server via TCP / IP socket took 5 seconds, I found very time consuming. So I wanted the ISO template which is very small and fast bytes.

brianrho, 2019-01-03:
Ah, that's true, every verification would mean an image transfer. You don't have any other means of identifying the user? Perhaps some unique ID that can be provided before their finger is checked.

thiagoanselmo, 2019-01-03:
You may even think of something, but do not become flexible and versatile.

Do you know of any readers who extract the template in ISO 19794-2 or ANSI format?

brianrho, 2019-01-03:
I don't know of any such readers though you can probably find them online. 
I worked on PC-side matching for these modules a while back and while it's not been well tested, I got good consistent results so far. It's all in C and you'll need to link in a DLL. I've uploaded it all here: https://github.com/brianrho/fpmatch

thiagoanselmo, 2019-01-03:
I have found some readers in China who do an ISO extraction. Thanks for sharing your code, I saw that you use windows, where can I find .dll for Linux?

brianrho, 2019-01-03:
I didnt find any Linux equivalent in the SDK, you'll have to search online for ways of using a DLL in Linux. Or you could get another reader that speaks ISO