brianrho/FPM #10: Transferring Image to Pc

sharad386, 2018-10-11:
Hi,
with the help of your library i am able to enroll, and get template but when i try to use "image to pc code"
it gives me following output
OUTPUT:

Waiting for a finger...
.
.
.
.
.
.
.
.
.
.
Image taken
Starting image stream...

and after this the execution stops ..can you help me out here i want to display the thump impression in my java application . Thank You

brianrho, 2018-10-11:
Did you wait for it to finish streaming the image? How long did you wait?

sharad386, 2018-10-12:
Almost 5-10 minutes. is this because my fingerprint sensor is different (the one will all red wires) and my template size is 768 bytes i have seen other fingerprint sensor that generates template of 512 byte!

sharad386, 2018-10-12:
[image]

sharad386, 2018-10-12:
after 2-3 seconds 

[image]

sharad386, 2018-10-12:
> after 2-3 seconds
> 
> [image]

and i get this error
[image]

brianrho, 2018-10-12:
I don't see any error and it looks like you're using the serial monitor to receive the image. You're supposed to run the python script simultaneously, else how will the image get decoded and saved to your PC? Upload the sketch and then run the script and follow the instructions. Let me know your results.

brianrho, 2018-10-16:
Took a look again and just spotted the error you spoke of, at the very end of the packet. Would like to see the rest of the error message. Does this happen every time you try the example? Even with the python script?

closed 2019-01-08 by brianrho