brianrho/FPM #11: R551 sensor does not get right image imgae_to_pc

baku-hatsu, 2018-12-16:
python output:
----------Extract Fingerprint Image------------

Options:
        Press 1 to enter serial port settings
        Press 2 to scan a fingerprint and save the image
        Press 3 to view help
        Press 4 to exit

>> 1

Enter Arduino serial port number: COM3
Enter serial port baud rate: 57600

Options:
        Press 1 to enter serial port settings
        Press 2 to scan a fingerprint and save the image
        Press 3 to view help
        Press 4 to exit

>> 2

Enter filename/path of output file (without extension): asd
SEND IMAGE TO PC test

[+]Got header
[+]Address: 0xFFFFFFFF
[+]PID: 0x7
[+]Length: 1
0
[+]Read complete

[+]Got header
[+]Address: 0xFFFFFFFF
[+]PID: 0x7
[+]Length: 17
0 0 0 0 0 1 F4 0 3 FF FF FF FF 0 2 0 6
[+]Read complete

[+]Got header
[+]Address: 0xFFFFFFFF
[+]PID: 0x7
[+]Length: 17
0 0 0 0 0 1 F4 0 3 FF FF FF FF 0 2 0 6
[+]Read complete
Found fingerprint sensor!
Capacity: 500
Packet length: 128

[+]Got header
[+]Address: 0xFFFFFFFF
[+]PID: 0x7
[+]Length: 1
0
[+]Read complete

[+]Got header
[+]Address: 0xFFFFFFFF
[+]PID: 0x7
[+]Length: 17
0 0 0 0 0 1 F4 0 3 FF FF FF FF 0 2 0 6
[+]Read complete
Packet length set to 128 bytes
Waiting for a finger...

[+]Got header
[+]Address: 0xFFFFFFFF
[+]PID: 0x7
[+]Length: 1
2
[+]Read complete

[+]Got header
[+]Address: 0xFFFFFFFF
[+]PID: 0x7
[+]Length: 1
0
[+]Read complete
Image taken

[+]Got header
[+]Address: 0xFFFFFFFF
[+]PID: 0x7
[+]Length: 1
0
[+]Read complete
Starting image stream...
Image saved as asd.bmp
FF FF FF FF FF FF FF FF FF FF FF FF FF EF 1 FF FF FF FF 2  0 82 FF FF FF FF

Picture is this:

[image]

Arduino seriam monitor shows this:
[+]Got header
[+]Address: 0xFFFFFFFF
[+]PID: 0x2
[+]Length: 128
⸮FF ⸮FF ⸮FF ⸮FF ⸮FF ⸮FF ⸮FF ⸮FF ⸮FF ⸮FF ⸮FF ⸮FF ⸮FF ⸮FF ⸮FF ⸮FF ⸮FF ⸮FF ⸮FF ⸮FF ⸮FF ⸮FF ⸮FF ⸮FF ⸮FF ⸮FF ⸮FF ⸮EF 1 ⸮FF ⸮FF ⸮FF ⸮FF 2 
Read data failed: -1

Error receiving packet 67

brianrho, 2018-12-16:
No, I said to provide a screenshot of the python console itself after the image is done being transferred. Dont paste the text. Upload an image of the console.

baku-hatsu, 2018-12-16:
When i turn off debuging the python exe doesnt work:
[image]

With debugging is open, i get this:
[image]

brianrho, 2018-12-16:
Okay. So this is the latest version of the library, right? And in the first case, you get "Timeout!" after waiting for a few seconds?

brianrho, 2018-12-16:
The "Timeout!" doesn't show immediately the image stream begins?

baku-hatsu, 2018-12-16:
Yes it dosnt show immediately, it takes like 10 or little more seconds. And this is latest version of library, with R551 uncommented line

brianrho, 2018-12-16:
It seems your module has a different image size of 192x192. I've pushed a change to the python script. Assign True to FPM_R551_MODULE at the top and run it. Let me know how it works out

brianrho, 2018-12-16:
It's important that debugging is turned off for this sketch because the debug messages go through the same Serial port as the image stream, so all those debug strings end up getting written to the bmp file, which isnt what you want.

baku-hatsu, 2018-12-16:
I get same error:
[image]
Did you recompiled python exe or it read from that file?

brianrho, 2018-12-16:
No I haven't rebuilt the exe. Can you run the python script?

baku-hatsu, 2018-12-16:
I edite phyton file, but i dont see changes, so i assume your changes is only on file, not on .exe, to build exe from python file i cant yet, i had lots of trouble with python on Windows 10, would be good if you could build it

baku-hatsu, 2018-12-16:
According to documentation image buffer size can be 192x192 with depth of 8
[image]

brianrho, 2018-12-16:
Yeah, made a mistake. In the python code, change the WIDTH = 96 to 192.

baku-hatsu, 2018-12-16:
I tried that myself, i still get error. Console looks like this:
[image]
Image looks like this:
[image]

brianrho, 2018-12-16:
That looks like the very last packet. Can you try running it again? I want to see if the failure is consistent

brianrho, 2018-12-16:
I think you didnt change this line `FPM_R551_MODULE = False` to `True` like I said. The arduino ends up waiting forever for another packet, when the image stream is already completed.

baku-hatsu, 2018-12-16:
Two tries out of five did not gave error message, however rest of the tries gave same error message. All scanned images was same size.
Code is:
[image]

brianrho, 2018-12-16:
What is the exact size? Check the properties of the generated image and send a screenshot of that. Also a screenshot of the console output when there was no error.

baku-hatsu, 2018-12-16:
Scanned image size is 192x192 pixels 
Console look like this:
port is COM3 and bound is 57600
[image]

baku-hatsu, 2018-12-16:
As well im using python -m pip install pyserial as for serial import for python, that maybe be other kind of serial?

brianrho, 2018-12-16:
No, its fine. Run the sketch, but use the serial monitor instead.

brianrho, 2018-12-16:
Looks like the problem isnt with the script, its with the module not sending FPM_ENDDATAPACKET to show that its done.

brianrho, 2018-12-16:
You haven't made any changes to the image_to_pc sketch, right?

baku-hatsu, 2018-12-16:
No, i havent, only to FPM.h library, where i uncommented one line of code 
/* R551 is different in a few ways (no high-speed search, off-by-one template indices)
   uncomment this line if you have one of those */
#define FPM_R551_MODULE

brianrho, 2018-12-16:
Okay. Comment out line 663 `outStream->write(byte);` in FPM.cpp. Enable debugging. Then run the sketch with the serial monitor alone. Let me see what is printed in the last few packets of the image, before the error shows up.

brianrho, 2018-12-16:
The FFs arent a problem, they just show that most of the image is white pixels as expected. 
Comment out these debug lines in the code: 
```
[+]Got header
[+]Address: 0xFFFFFFFF
```
As well as line 670: `FPM_DEBUG_HEX(byte); FPM_DEBUG_PRINT(" ");`

I want only the PID and length to be printed.

baku-hatsu, 2018-12-16:
`[+]PID: 0x2
[+]Length: 128
Read data failed: -1

Error receiving packet 144

`

brianrho, 2018-12-16:
Run it a few more times, check that it gives the same result and stops at the same packet

baku-hatsu, 2018-12-16:
Yes its always same packet

brianrho, 2018-12-16:
And this is the very last packet always:
```
[+]PID: 0x2
[+]Length: 128
```

Can you check if any packet has a PID of 0x8?

baku-hatsu, 2018-12-16:
There is no packet with PID 0x8

brianrho, 2018-12-16:
At line 662 in FPM.cpp, change this: 
```
if (outStream != NULL) {
      outStream->write(byte);
}
```
to
```
if (outStream != NULL) {
      outStream->write(byte);
      chksum += byte;
}
```
and this at 685:
`if (outStream == NULL && to_check != chksum) `
to:
`if (to_check != chksum)`

brianrho, 2018-12-16:
Let me know if it prints "Wrong chksum" for any packet.

brianrho, 2018-12-16:
`outStream->write(byte);` should still be commented out as before.

baku-hatsu, 2018-12-16:
It prints that for every stream packet

brianrho, 2018-12-16:
What of the PID and length? There was no wrong chksum in the earlier commands, right?

baku-hatsu, 2018-12-16:
[+]Wrong chksum
[+]PID: 0x2
[+]Length: 128

[+]Wrong chksum

Read data failed: -1

Error receiving packet 0

brianrho, 2018-12-16:
You mean, there was no "wrong chksum" before the stream began, but then after that, each packet has a wrong chksum? ALL the packets with wrong chksums have a PID of 0x2?

baku-hatsu, 2018-12-16:
Yes

baku-hatsu, 2018-12-16:
https://pastebin.com/Xcpz16Vv the serial monitor
https://pastebin.com/vZpz5D75 fpm.cpp file

brianrho, 2018-12-16:
Have you restarted the module, since you started using it to get images?

brianrho, 2018-12-16:
The packet length setting is being ignored for some reason and it could be because it hasn't taken hold yet.

baku-hatsu, 2018-12-16:
Probably not, how do i reset it? using reset pin or just unplug?

brianrho, 2018-12-16:
Just unplug from power and try again.

baku-hatsu, 2018-12-16:
There is same error

brianrho, 2018-12-16:
Okay then. Lets reset the packet length to 32 and restart. Change this line in the sketch:

`uint8_t value = FPM_PLEN_128;`

to:

`uint8_t value = FPM_PLEN_32;`

Restart the module and try the sketch again. Update the serial monitor dump at the same pastebin link.

brianrho, 2018-12-16:
Remove the header and address debug messages too

baku-hatsu, 2018-12-16:
https://pastebin.com/SRYQ6X1J new pastebin, because i cant edit old one

brianrho, 2018-12-16:
How are you powering the module?

brianrho, 2018-12-16:
The Arduino 3.3V pin?

brianrho, 2018-12-16:
Yeah try powering the sensor with the RPi 3.3V pin, no need to connect anything else to the RPi, just power it. Run the sketch again and update the dump. You can update it at the bottom

brianrho, 2018-12-16:
Sorry, you can edit it at the top of the text window. Just use Find in your browser to search for "edit"

baku-hatsu, 2018-12-16:
Apparently i have to be logged in to be able to edit so i just rather paste new ones
https://pastebin.com/5hiCpBND

brianrho, 2018-12-16:
Okay. Run it again. Seems the very first image packet has a correct checksum but the ones that follow are wrong.

baku-hatsu, 2018-12-16:
https://pastebin.com/tVU2i7zR

this is what i got now

brianrho, 2018-12-16:
Okay comment out the address, PID and length messages. Only the image bytes should be printed and "wrong chksum"

baku-hatsu, 2018-12-16:
https://pastebin.com/TxLStMxC

brianrho, 2018-12-16:
Run the "set_parameters" example. It will change the baud rate to 9600. Then run the image_to_pc again with SoftwareSerial baud rate of 9600.

brianrho, 2018-12-16:
Remember to comment out the Address debug messages.

baku-hatsu, 2018-12-16:
Apparently for some reason Arduino does not recognize fingerprint anymore

brianrho, 2018-12-16:
Did you update this line `fserial.begin(57600);` to `fserial.begin(9600);`?

[55 more replies]

closed 2019-05-25 by brianrho