brianrho/FPM #13: R551 Did not find fingerprint sensor :(

MaazSk, 2019-02-04:
Hie,
I have purchased a R551 sensor it was working nice when I uploaded some of the codes and was able to save the fingerprints and access those templates, but all of the sudden it stopped working and I am getting the message "Did not find fingerprint sensor :(" on the serial monitor.
If anyone knows how to solve the issue please help me out.
Thanks in advance

brianrho, 2019-02-05:
Did you run the set_parameters example? If you did, it may have changed your baud rate to 9600. So change your SoftSerial baud rate in the enroll example and try again

MaazSk, 2019-02-05:
I tried changing the baud rate also but I am getting the same message "Did not find fingerprint sensor". I have one more sensor it is working good on the same piece of code does this mean my sensor is broke or is having some problems and it will not work if yes can you suggest me how to fix it @brianrho 
Thank You!

mohammadhasanzadeh, 2019-02-05:
Unfortunately, I have the same problem on R308 fingerprint sensor, but the same code works on ZFM-60!
Also, I tried to change baud rate to `9600`, but problem not fixed.

MaazSk, 2019-02-07:
The problem is solved actually I had uploaded the change password code which changed the password to "0x12345678", now I have changed the password again to "0" and the sensor is working fine 
Thank You! for your help @brianrho

closed 2019-02-07 by MaazSk

brianrho, 2019-02-07:
Ah, glad to hear it worked out, will have to add a warning to the sketch.