brianrho/FPM #15: Did not find fingerprint sensor error on R308

mohammadhasanzadeh, 2019-02-06:
Unfortunately, I have the problem on R308 fingerprint sensor, but the same code works on ZFM-60!
Also, I tried to change baud rate to `9600` and `57600`, but problem not fixed.

MaazSk, 2019-02-07:
@mohammadhasanzadeh  Have you uploaded the change password code?
If yes just change the password to the default one and it will work fine, else just try to swap the Rx and Tx pins there are times when these 2 pins are not connected properly we will get the same issue.
Just let us know if the issue is solved

brianrho, 2019-02-11:
@mohammadhasanzadeh did @MaazSk's suggestion fix your problem? Will be closing the issue if this is done

mohammadhasanzadeh, 2019-02-11:
> @mohammadhasanzadeh  Have you uploaded the change password code?
> If yes just change the password to the default one and it will work fine, else just try to swap the Rx and Tx pins there are times [...]

No, I didn't change password, also I swap several times Tx and Rx pins.
is there any way to reset fingerprint in hardware mode?

MaazSk, 2019-02-11:
@mohammadhasanzadeh There is a reset pin which is pin 7 but I don't know how to reset the hardware. If there is an option for hardware reset please let me know and if the issue is solved then also let me know. I am working on the same module if I come across anything will let you know

Thanks and Regards

brianrho, 2019-02-11:
I don't know of any hardware reset pin, at least not with the sensors I've worked with. Most times when sensor communication fails, its usually because of a wrong baud rate or password. What is your MCU or dev board?

mohammadhasanzadeh, 2019-02-11:
I did not find any solution for the reset fingerprint sensor in the datasheet.
I use `wemos d1 mini lite`.

brianrho, 2019-02-11:
What pins are you using? What's your `SoftwareSerial` declaration like?

mohammadhasanzadeh, 2019-02-11:
I use `12` for Rx and `14` for Tx:
`SoftwareSerial finger_print_serial(12, 14);`

MaazSk, 2019-02-11:
Can you please send us the code

MaazSk, 2019-02-11:
@mohammadhasanzadeh  have you installed the software serial library for wemos or NodeMCU boards there is a different Software Serial library for these boards the arduino library is not compatible, please make this change and let us know 
Thanks and Regards
Maaz Shaikh

brianrho, 2019-02-11:
- This is the mapping you've been using for your wiring?
D6 = GPIO12 = Sensor TX (green usually)
D5 = GPIO14 = Sensor RX (yellow usually)

- How have you been powering the sensor? With 5V?

- I've assumed so far that you're running the examples but if you're not, then you should try the builtin `enroll` example. Try changing to `finger.begin(0x12345678)` if it doesn't work at first.

mohammadhasanzadeh, 2019-02-11:
Is very strange problem! I upload the Adafruit example:
[](https://github.com/adafruit/Adafruit-Fingerprint-Sensor-Library/blob/master/examples/fingerprint/fingerprint.ino)
and without any change in pins, it works correctly!

brianrho, 2019-02-11:
Let me see the sketch you used before this

mohammadhasanzadeh, 2019-02-11:
main.cpp:
https://pastebin.com/qjb23QDC

finger_print.cpp:
https://pastebin.com/QHhDtxeP

brianrho, 2019-02-11:
This isn't the example sketch I said you should try

brianrho, 2019-02-11:
Will be closing this, if there are no problems with the examples

mohammadhasanzadeh, 2019-02-11:
I have the same problem with `enroll` example.

brianrho, 2019-02-11:
Send the exact sketch with pastebin

brianrho, 2019-02-11:
As well as what you see in the serial monitor

mohammadhasanzadeh, 2019-02-11:
enroll sketch:
https://pastebin.com/QiAwbYsY
Serial monitor output:
```
ENROLL test
Did not find fingerprint sensor :(
```

brianrho, 2019-02-11:
The pins in your sketch are reversed, compared to what you sent earlier

mohammadhasanzadeh, 2019-02-11:
Yes, I changed pins for equal to:
```
D6 = GPIO12 = Sensor TX (green usually)
D5 = GPIO14 = Sensor RX (yellow usually)
```

also I tested both:
`SoftwareSerial fserial(12, 14);`
and 
`SoftwareSerial fserial(14, 12);`
but both are wrong!

brianrho, 2019-02-11:
Did any of the FPM examples ever work with this sensor, from the beginning?

mohammadhasanzadeh, 2019-02-11:
No, not any example works with this sensor, but `Adafruit` examples works with R308…!

brianrho, 2019-02-11:
Uncomment FPM_DEBUG in FPM.h and run the enroll example again with the right pins, let me see everything you get from Serial

mohammadhasanzadeh, 2019-02-11:
The following lines printed in debug mode:

```
�Hl�H�C�d�␇X␡�ENROLL test

[+]Got header
[+]Address: 0xFFFFFFFF
[+]PID: 0x7
[+]Length: 1
0
[+]Read complete

[+]Got header
[+]Address: 0xFFFFFFFF
[+]PID: 0x7
[+]Length: 1
1
[+]Read complete
Did not find fingerprint sensor :(
```

brianrho, 2019-02-11:
The problem is, the initial handshake passes with return code 0 as expected, but the subsequent readParams() call returns 1, as you can see from the Serial output. 
From what I've found in the datasheet and user reviews, it seems the R308 doesn't support the ReadSysParam command, unlike most other sensors in this family. This means the library can't automatically determine the sensor capacity, packet length, etc, which are integral to the library logic at the moment. So I'd recommend you either use the Adafruit library or get another   more compatible sensor like the FPM10 or ZFM60

closed 2019-02-11 by brianrho

mohammadhasanzadeh, 2019-02-17:
In `FPM::readParams` method I hard-coded the following lines:
`   sys_params.status_reg = 0x04;
    sys_params.system_id = 0x0009;
    sys_params.capacity = 500;
    sys_params.security_level = FPM_FRR_5;
    sys_params.device_addr = 0xFF;
    sys_params.packet_len = FPM_PLEN_128;
    sys_params.baud_rate = FPM_BAUD_57600;`
After that, the fingerprint sensor also worked properly and also features that I needed, such as `Enroll finger`, `Fast search`, `Get free id`, `Read template`, `Store template`, `delete template` and `Clear database` all of them worked correctly.
Just in `Store template`,  occurs `TIMEOUT` error randomly.

MaazSk, 2019-02-17:
Great nice to hear that, I am also having some problem in templates example working on the same

mohammadhasanzadeh, 2019-02-17:
Of course after some tries `Store template` finally works.

brianrho, 2019-02-17:
Good to hear. I'll consider adding the option of manually supplying the parameters

MaazSk, 2019-02-17:
@mohammadhasanzadeh  what changes have you made for the templates example?

mohammadhasanzadeh, 2019-02-17:
> @mohammadhasanzadeh what changes have you made for the templates example?

I did not exactly test `templates` example, I tested my own code.
But for running `templates` example, I think you just need change `FPM::readParams` in `FPM.cpp` to:
```
int16_t FPM::readParams(FPM_System_Params * user_params) {
    // buffer[0] = FPM_READSYSPARAM;
    
	// writePacket(FPM_COMMANDPACKET, buffer, 1);
    // uint8_t confirm_code = 0;
    // int16_t len = read_ack_get_response(&confirm_code);
    
    // if (len < 0)
    //     return len;
    
    // if (confirm_code != FPM_OK)
    //     return confirm_code;
    
    // if (len != 16)
    //     return FPM_READ_ERROR;
    
    // memcpy(&sys_params, &buffer[1], 16);
    // reverse_bytes(&sys_params.status_reg, 2);
    // reverse_bytes(&sys_params.system_id, 2);
    // reverse_bytes(&sys_params.capacity, 2);
    // reverse_bytes(&sys_params.security_level, 2);
    // reverse_bytes(&sys_params.device_addr, 4);
    // reverse_bytes(&sys_params.packet_len, 2);
    // reverse_bytes(&sys_params.baud_rate, 2);
    sys_params.status_reg = 0x04;
    sys_params.system_id = 0x0009;
    sys_params.capacity = 500;
    sys_params.security_level = FPM_FRR_5;
    sys_params.device_addr = 0xFF;
    sys_params.packet_len = FPM_PLEN_128;
    sys_params.baud_rate = FPM_BAUD_57600;
    
    if (user_params != NULL)
        memcpy(user_params, &sys_params, 16);
    
    return FPM_OK;
}
```

mohammadhasanzadeh, 2019-02-18:
@brianrho `Timeout` error still occur after call `storeModel` randomly.
in what cases may occurs `Timeout` state?

brianrho, 2019-02-18:
Usually whenever it takes too long to get a response because of wrong baud rates, passwords, etc, hardly ever happens but you've made changes so could be anything

brianrho, 2019-03-09:
I've made changes to support the R308 as well as updated the debug system. You can check with your R308 to see if it works. Set the debug level to 2 so you can see everything that happens, could help track down the source of the timeouts.

JavadMohammdi, 2021-05-02:
> > @mohammadhasanzadeh what changes have you made for the templates example?
> 
> I did not exactly test `templates` example, I tested my own code.
> But for running `templates` example, I think you just need change `FPM::readParams` in `FPM.cpp` to:
> 
> ```
> int16_t FPM::readParams(FPM_System_Params * user_params) {
>     // buffer[0] = FPM_READSYSPARAM;
>     
> 	// writePacket(FPM_COMMANDPACKET, buffer, 1);
>     // uint8_t confirm_code = 0;
>     // int16_t len = read_ack_get_response(&confirm_code);
>     
>     // if (len < 0)
>     //     return len;
>     
>     // if (confirm_code != FPM_OK)
>     //     return confirm_code;
>     
>     // if (len != 16)
>     //     return FPM_READ_ERROR;
>     
>     // memcpy(&sys_params, &buffer[1], 16);
>     // reverse_bytes(&sys_params.status_reg, 2);
>     // reverse_bytes(&sys_params.system_id, 2);
>     // reverse_bytes(&sys_params.capacity, 2);
>     // reverse_bytes(&sys_params.security_level, 2);
>     // reverse_bytes(&sys_params.device_addr, 4);
>     // reverse_bytes(&sys_params.packet_len, 2);
>     // reverse_bytes(&sys_params.baud_rate, 2);
>     sys_params.status_reg = 0x04;
>     sys_params.system_id = 0x0009;
>     sys_params.capacity = 500;
>     sys_params.security_level = FPM_FRR_5;
>     sys_params.device_addr = 0xFF;
>     sys_params.packet_len = FPM_PLEN_128;
>     sys_params.baud_rate = FPM_BAUD_57600;
>     
>     if (user_params != NULL)
>         memcpy(user_params, &sys_params, 16);
>     
>     return FPM_OK;
> }
> ```
> 
> R308 fingerprint sensor just works on `57600` baud rate, so it's important you ensure your software serial works on `57600` baud rate.

Hi, I wanted to register my r308 fingerprint in Arduino, which gives an error and your problem can help me

JavadMohammdi, 2021-05-02:
> I did not exactly test `templates` example, I tested my own code.
> But for running `templates` example, I think you just need change `FPM::readParams` in `FPM.cpp` to:
>
> ```
> int16_t [...]

Can you send me the code with which you could register your fingerprint on r308?

brianrho, 2021-05-04:
@JavadMohammdi As explained in the README, there's an [example](https://github.com/brianrho/FPM/tree/master/examples/R308_seach_database) already that shows how to setup your R308 to work with this library. You can easily port/transfer the `setup()` in that example to the other examples like `enroll` and `templates`.