brianrho/FPM #16: Unable to run templates example in R551

MaazSk, 2019-02-13:
Hie, 
I tried running the templates example but was not able to successfully run it I am getting the following message please help me resolve it. 

TEMPLATES test
Found fingerprint sensor!
Capacity: 500
Packet length: 128
Enter the template ID # you want to get...
Template 1 loaded
Error receiving packet

Below is the text which I am getting after I have removed the debug comment

TEMPLATES test

[+]Got header
[+]Address: 0xFFFFFFFF
[+]PID: 0x7
[+]Length: 1
0 
[+]Read complete

[+]Got header
[+]Address: 0xFFFFFFFF
[+]PID: 0x7
[+]Length: 17
0 0 0 0 0 1 F4 0 3 FF FF FF FF 0 2 0 6 
[+]Read complete

[+]Got header
[+]Address: 0xFFFFFFFF
[+]PID: 0x7
[+]Length: 17
0 0 0 0 0 1 F4 0 3 FF FF FF FF 0 2 0 6 
[+]Read complete
Found fingerprint sensor!
Capacity: 500
Packet length: 128
Enter the template ID # you want to get...

[+]Got header
[+]Address: 0xFFFFFFFF
[+]PID: 0x7
[+]Length: 1
0 
[+]Read complete
Template 1 loaded

[+]Got header
[+]Address: 0xFFFFFFFF
[+]PID: 0x7
[+]Length: 1
0 
[+]Read complete

[+]Got header
[+]Address: 0xFFFFFFFF
[+]PID: 0x2
[+]Length: 128
4 B 6B 0 B2 9A 85 64 C2 B 42 E8 D AB 85 E0 40 DA 3B 7 1A 43 86 6A 41 DC 79 DF 39 33 87 66 41 A 78 E7 91 63 88 63 C2 8 38 E8 CE 6B 87 D9 68 EF 1 0 0 0 0 0 EF 1 FF FF 32 0 0 
Read data failed: -1
Error receiving packet
Enter the template ID # you want to get...

Thanks and Regards 
Maaz Shaikh.

brianrho, 2019-02-13:
Why did you change it to 496? Does it work with the default 768 bytes buffer?

brianrho, 2019-02-13:
Run the example as it is, change ONLY the pins and disable debug

MaazSk, 2019-02-13:
I am getting the same error. Running the example code provided in the library

TEMPLATES test
Found fingerprint sensor!
Capacity: 500
Packet length: 128
Enter the template ID # you want to get...
Template 1 loaded
Error receiving packet
Enter the template ID # you want to get...

MaazSk, 2019-02-13:
It is not working with with the default 768 neither it is working with 496

brianrho, 2019-02-13:
Edit the sketch and add a Serial.println(count) after it prints the error message you're getting. Run it with 768 bytes as usual and let me see

brianrho, 2019-02-13:
Have you tried it? What did you get?

MaazSk, 2019-02-14:
Hi briarho,
I'll try it tomorrow as I don't have the access to my laptop and
fingerprint sensor today I'll just check it by tomorrow and will let you
Thanks and regards

MaazSk, 2019-02-15:
Below is the output that I receive when I print count

TEMPLATES test
Found fingerprint sensor!
Capacity: 500
Packet length: 128
Enter the template ID # you want to get...
Template 1 loaded
1
2
3
4
Error receiving packet
Enter the template ID # you want to get...
Template 0 loaded
1
2
3
4
Error receiving packet
Enter the template ID # you want to get...
Template 2 loaded
1
2
3
4
Error receiving packet
Enter the template ID # you want to get...

brianrho, 2019-02-15:
Where did you place the print?

MaazSk, 2019-02-15:
Below is the code where I have place the print command

` while (true) {
        bool ret = finger.readRaw(FPM_OUTPUT_TO_BUFFER, buffer + pos, &read_finished, &readlen);
        if (ret) {

            count++;

            Serial.println(count);   //// Here I am printing

            pos += readlen;

            readlen = buff_sz - pos;

            if (read_finished)

                break;
        }
        else {

            Serial.println("Error receiving packet");

            return 0;
        }
        yield();`

brianrho, 2019-02-15:
Change this IF statement at line 305 at FPM.cpp:

 ```
   if (len != chunk_sz) {
        FPM_DEBUG_PRINT("Read data failed: ");
        FPM_DEBUG_PRINTLN(len);
        return false;
    }
```
to:

 ```
   if (len != chunk_sz) {
        FPM_DEBUG_PRINT("Read data failed: ");
        FPM_DEBUG_PRINTLN(len);
        Serial.print("Len: "); Serial.println(len);
        return false;
    }
```
and try again.

MaazSk, 2019-02-16:
I have done the suggested changes but still I am not able to get the desired output. Below is the message which I receive on the Serial Monitor.

TEMPLATES test
Found fingerprint sensor!
Capacity: 500
Packet length: 128
Enter the template ID # you want to get...
Template 1 loaded
1
2
3
Len: 54
Error receiving packet
Enter the template ID # you want to get...
Template 2 loaded
1
2
3
Len: 54
Error receiving packet
Enter the template ID # you want to get...

brianrho, 2019-02-16:
The changes are to find out what's going on, not to fix it.
What are the current baud rates for Serial and SoftwareSerial? What is the buffer size? And, did this example ever work before with this sensor?

MaazSk, 2019-02-16:
The baud rate is 9600 for serial and 57600 for software serial buffer size is 768 and tried with 496 also. No, this example did not worked with this sensor

brianrho, 2019-02-16:
Stick with the 768 byte buffer for now, don't want it running out of space. Which buffer size gave the results you sent earlier?

MaazSk, 2019-02-16:
Okay I will stick to 768 only. Can you please a tell me about which results you are talking about.

brianrho, 2019-02-16:
Just rerun the most recent changes I sent you and post the results

MaazSk, 2019-02-16:
Okay 
TEMPLATES test
Found fingerprint sensor!
Capacity: 500
Packet length: 128
Enter the template ID # you want to get...
Template 1 loaded
1
2
3
4
Len: 54
Error receiving packet
Enter the template ID # you want to get...
Template 0 loaded
1
2
3
4
Len: 54
Error receiving packet
Enter the template ID # you want to get...

brianrho, 2019-02-16:
Okay, comment out that whole IF block and run the sketch again

MaazSk, 2019-02-16:
Okay wait

MaazSk, 2019-02-16:
This is what I am getting the length is constant

TEMPLATES test
Found fingerprint sensor!
Capacity: 500
Packet length: 128
Enter the template ID # you want to get...
Template 0 loaded
Len: 54
Error receiving packet
Enter the template ID # you want to get...
Template 1 loaded
Len: 54
Error receiving packet
Enter the template ID # you want to get...
Template 2 loaded
Len: 54
Error receiving packet
Enter the template ID # you want to get...
Template 3 loaded
Len: 54
Error receiving packet
Enter the template ID # you want to get...
Template 4 loaded
Len: 54
Error receiving packet
Enter the template ID # you want to get...
Template 5 loaded
Len: 54
Error receiving packet
Enter the template ID # you want to get...
Template 6 loaded
Len: 54
Error receiving packet
Enter the template ID # you want to get...
Template 7 loaded
Len: 54
Error receiving packet
Enter the template ID # you want to get...
Template 8 loaded
Len: 54
Error receiving packet
Enter the template ID # you want to get...
Template 9 loaded
Len: 54
Error receiving packet
Enter the template ID # you want to get...

brianrho, 2019-02-16:
No I said you should comment out the whole IF block that contains that statement. In FPM.cpp

MaazSk, 2019-02-16:
Ohh okay

MaazSk, 2019-02-16:
The template size is constant 640 

TEMPLATES test
Found fingerprint sensor!
Capacity: 500
Packet length: 128
Enter the template ID # you want to get...
Template 0 loaded
Printing template:
---------------------------------------------
4 11 6B 0 5 63 85 5A 6 7A 3 E7 2D 73 86 4E 
45 AA 47 D8 75 3B 88 BB 43 1A 38 F 66 73 87 4B 
4 A9 45 E0 8D F3 86 C3 5 49 76 17 5E 6C 86 D4 
C4 2C 74 F 71 2C 87 57 7 BA 47 F8 29 12 87 3C 
1 F7 43 0 66 A2 84 50 C8 FA FF E7 8A 22 87 B7 
C2 29 3E 8 AE 7A 87 40 82 FB 45 E8 9A 1B 89 41 
C3 DA 45 F8 EE 8B 88 47 3 CB 45 E0 71 54 88 C1 
2 FA FB 1F EE DB 82 E1 2 FB 79 F8 C1 A4 87 51 
87 CA C9 F8 32 75 87 5B 4A FB 7F F8 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 41 45 78 33 34 34 22 73 13 2F 24 65 
12 24 76 34 34 5 8F 27 82 96 62 71 18 F2 F7 41 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
D 6B 38 52 42 81 EA 91 78 4D 31 A1 DC A1 34 23 
53 C1 8C A1 20 45 4 17 E0 C1 F6 59 31 B0 A9 91 
8A 15 40 80 CC A1 46 5C 11 B0 B8 A0 7C 43 63 71 
11 B0 AF 73 30 35 8D 90 8F 32 31 30 56 B0 A2 6E 
1 21 55 A1 8F 67 11 32 9D A0 DC 44 10 6B 4 A0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 C4 6D 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
--------------------------------------------
640 bytes read.
Template 0 deleted
Enter the template's new ID...
Unknown error
Enter the template ID # you want to get...
Template 2 loaded
Printing template:
---------------------------------------------
4 13 57 0 2D 2 85 84 84 46 FE 7 6A 1A 89 27 
5 BA 4D F8 C2 52 85 89 A A7 74 A0 FD 72 85 8E 
52 CA 74 A0 F5 3A 87 3A CA 77 89 E6 19 9B 84 87 
5A F8 3E C0 31 CB 84 6A 6C F9 7E D9 72 6B 85 5F 
A 3B 47 DA AD F3 86 52 7 67 C5 EE 9A 8A 89 9D 
8 FA FF 27 ED 52 89 AE 86 4B F2 EF AE 73 89 BF 
5 2D F6 EF 1A C 85 66 C5 97 7F FF 32 74 85 61 
85 A9 7F F8 C2 51 87 93 81 2A FE 7 A F2 85 8D 
83 37 F8 7 4A 62 84 7D 4 55 FE EF 2D 2B 84 73 
4B E6 BE E0 62 83 84 6D 8E 37 3F E1 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 43 F4 2 89 32 6F 36 54 CF 77 31 61 
62 78 33 18 87 72 67 34 92 19 91 1F 58 43 5F 67 
49 9F 22 F 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
E 57 41 54 43 45 B3 72 F9 0 44 70 D8 A4 20 37 
34 53 12 71 3A 17 22 51 D0 72 3C 13 1 73 E2 B2 
54 8 63 E1 ED 70 9D 22 0 50 8A 9B 74 1A 1 D5 
88 86 62 2A 10 2 88 61 19 5A 30 65 BB B1 77 1D 
16 C3 AD 73 F 3 61 56 13 A1 B6 1D 4 C5 9C 62 
0 32 68 20 8E A2 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 E 79 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
--------------------------------------------
640 bytes read.
Template 2 deleted
Enter the template's new ID...
Unknown error
Enter the template ID # you want to get...
Invalid template
Enter the template ID # you want to get...
Invalid template
Enter the template ID # you want to get...
Invalid template
Enter the template ID # you want to get...
Template 4 loaded
Printing template:
---------------------------------------------
4 C 49 0 29 13 87 26 82 F6 3 F8 39 B3 84 91 
85 34 FA 7 9D 7B 86 9E 84 45 78 17 AD 33 86 2B 
84 B5 7 E0 AD 33 85 97 85 54 78 30 91 3B 89 2D 
84 F8 43 FF 71 AC 88 AE 81 F8 F5 F 69 73 83 87 
84 54 BE E8 DA 63 83 87 49 98 7E E0 7D A4 86 B1 
8A 78 74 2F C6 34 88 47 C9 88 7F FF BD DC 86 4F 
48 68 BF EE 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 55 6A 14 B6 27 56 51 C6 B8 75 44 FF 
4F F9 B5 35 36 F 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
6 49 6F 20 14 42 95 51 E9 11 41 92 DD 51 EE 0 
44 2B D5 82 BC 8 11 54 E7 41 EE 22 56 E2 9E 82 
8B 6B 4 65 AE 80 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 9F 49 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
--------------------------------------------
640 bytes read.
Template 4 deleted
Enter the template's new ID...
Unknown error

brianrho, 2019-02-16:
Okay, seems so. When it asked for a new ID, what did you enter? Try entering an ID you've never used like 200 or something

MaazSk, 2019-02-16:
okay

MaazSk, 2019-02-16:
I am getting the same message

brianrho, 2019-02-16:
Are you sure your sensor is an R551? Not only is the template size different from the 768 expected, but it also seems it doesn't support the DOWNCHAR command which is necessary to upload templates. Do you still have the product link?

MaazSk, 2019-02-16:
I don't have the product link right now but I am sure it is R551, and yes the template size is different than other sensor and to that which is mentioned in the datasheet I'll just dig in and find the product link and will provide you till tomorrow 
Thank You!

brianrho, 2019-02-16:
Sure. You can try a few things later like:

- Increase the buffer size to 1024 and see if there are any differences

- Change the value in this line in the uploadModel() function in FPM.cpp:
`buffer[1] = 0x01;` to 0x02 and 0x03, see if anything changes.

MaazSk, 2019-02-16:
Okay I will try and let you know

MaazSk, 2019-02-16:
If I increase the buffer size then the arduino is not working because I have only 1024 bytes on the arduino uno and I have changed it to 0x01, 0x02, there is no change in the output

brianrho, 2019-02-16:
No the Uno has 2k bytes for RAM. How many bytes does it report are left after compilation?

Also did you try 0x03 like I said?

MaazSk, 2019-02-16:
Yes I tried 0x03 also as u said. Let me check the bytes and will let you
know

MaazSk, 2019-02-16:
Uno has only 1 Kb of RAM

brianrho, 2019-02-16:
The Uno has 2k of SRAM: https://www.arduino.cc/en/tutorial/memory

Place this in the sketch after it prints the "Unknown Error" in `move_template()`:

`Serial.println(p);`

I want to know the returned error code.

MaazSk, 2019-02-17:
This is what I get

Unknown Error 
-2

brianrho, 2019-02-17:
Make sure `buffer[1] = 0x01` always in `uploadModel()` in FPM.cpp. Then inside the function, place `Serial.prinltln(rc)` right inside the first IF block, before the function returns.

MaazSk, 2019-02-17:
This is the output after the suggested changes

TEMPLATES test
Found fingerprint sensor!
Capacity: 500
Packet length: 128
Enter the template ID # you want to get...
Invalid template
Enter the template ID # you want to get...
Template 2 loaded
Printing template:
---------------------------------------------
4 E 67 0 A6 52 85 DD 8A FA B1 F8 6D 1B 85 7B 
45 97 BF EE 8D 23 86 E3 44 39 76 0 9D 6B 85 75 
84 A9 7F EF AE 7B 88 D8 84 2A B4 E8 E9 A3 89 68 
C4 FB 7B FE 5 84 88 DF 83 B F6 DF 12 7C 86 6E 
2 BC 43 FF 32 84 85 E6 41 FC FB 17 65 9C 88 E7 
C2 DC FF D7 25 12 85 C3 25 EC BE F8 35 A 86 BE 
C AA 7E F8 D9 92 86 CD 87 88 76 10 11 2C 85 78 
82 B6 7F F8 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 63 13 4F 62 85 44 71 F5 45 44 23 A2 
21 23 16 49 42 73 D4 1F 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
A 67 77 2E 37 B2 99 81 7 4 1 B3 EC 91 14 1C 
60 E7 AC A1 EF 70 11 35 43 C0 CD 12 2 50 D5 B0 
A0 1B 5 E1 CA 91 64 1F 13 B2 AC 71 1C F 22 11 
C0 A2 16 22 2C E1 AD A3 70 1B 63 10 BD C0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 33 5F 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
--------------------------------------------
640 bytes read.
Template 2 deleted
Enter the template's new ID...

brianrho, 2019-02-17:
Okay, in the same FPM.cpp, replace this:

```
    /* wrong pkt id */
    if (pktid != FPM_ACKPACKET)
         return FPM_READ_ERROR;
```

with:

```
/* wrong pkt id */
    if (pktid != FPM_ACKPACKET) {
        Serial.print("PID: "); Serial.println(pktid);
        return FPM_READ_ERROR;
    }
```

MaazSk, 2019-02-17:
PID: 1
-2
Unknown Error
-2
This is the output

brianrho, 2019-02-17:
Well, that makes no sense at all. I'm afraid I can't do much more than this without having the sensor myself. And it seems theres isnt a lot more to be done anyways, nor is it worth it, given the disparities between the sensor's actual behaviour and whats described in its "datasheet". You'll have to either leave this functionality or get another module like the FPM10 or R307. I'll add another warning about this module to the readme, a good chunk of the messages here are a result of modules claiming to be the R551.

MaazSk, 2019-02-17:
Okay Thank you bro. can you just explain me a bit how the templates are moved from one ID to another I will carry on the research and if I found something that is helpful I will post for sure here so that you can make the necessary changes.
Again Thanks a lot for your help.

brianrho, 2019-02-17:
The code is mostly clear I think. After you've downloaded the template to your array, just follow the steps in move_template(). 

The DOWNCHAR command is first sent to the module to initiate the transfer of the template to the module. As part of the command, you supply the particular buffer 1 or 2, that is the temporary destination. The device must ACK the command (which is what has failed so far) before you can begin the actual transfer of the template with `writeRaw()`. After which you then store the template with the STORE command, passing in the new ID. The module internally transfers template from the internal buffer to the specified ID.

MaazSk, 2019-02-17:
Okay Thank You!

closed 2019-02-17 by brianrho