brianrho/FPM #17: R551 Trouble in templates.ino (SoftwareSerial buffer size and check answer length)

vvvait, 2019-02-19:
Hello. I use Wemos P1. I was able to run this example by increasing the buffer of the SoftwareSeries to 768 by default there is 64. 
`SoftwareSerial fserial(D1, D2, false, BUFF_SZ);`
It seems that the P551 writes to the buffer too quickly, and it overflows. Also had to comment out checking the length of the answer in FPM::readRaw
`/* check the length `
`    if (len != chunk_sz) {`
`        FPM_DEBUG_PRINT("Read data failed: ");`
`        FPM_DEBUG_PRINTLN(len);`
`        return false;`
`    } */`
because last packet was:
`[+]Got header`
`[+]Address: 0xFFFFFFFF`
`[+]PID: 0x8`
`[+]Length: 54`
`21 0 BE A4 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 19 64`
len (54) <> chunk_sz (from enum = 128)
There must be another length check for the last packet.

Thanks for the library, here more docs, for R551 too
Https://www.dropbox.com/sh/orprmb3bgb6lqb6/AACpiIXOF91R7-RQ9OkD4JXha?dl=0

brianrho, 2019-02-19:
Yeah, I'll be pushing changes today since I recently noticed some template sizes aren't multiples of the packet length. Is your template size 640 or 768? 

The 64 byte buffer of your SoftwareSerial should be enough, the actual problem is the enabled debug messages, they waste time when flushing especially, which makes the SoftwareSerial RX buffer overflow. Will add debug levels maybe so one can view errors without having to enable everything.

vvvait, 2019-02-19:
Template size 640 bytes.
Indeed, without a debug log, 64 bytes is enough

brianrho, 2019-02-19:
I've made the changes, try running the example again and see if it works. Disable debug and un-comment `FPM_R551_MODULE`

vvvait, 2019-02-19:
Buffer loading does not work. The documentation is missing a description of the DownChar process. In response to the uploadModel does not receive a confirmation, but an echo of the command

```
Enter the template's new ID...

[+]Got header
[+]Address: 0xFFFFFFFF
[+]PID: 0x1
[+]Length: 2
9 1
[+]Read complete
Unknown error -2
```
Attempts to ignore the error and start the download result in an error

```
[+]Got header
[+]Address: 0xFFFFFFFF
[+]PID: 0x7
[+]Length: 1
1
[+]Read complete
Comms error
```

brianrho, 2019-02-19:
Yes, that's what #16 was about, was closed just yesterday. The R551 seems to not support DOWNCHAR even if the final table in the datasheet does mention it. You'll have to get a different sensor if you want that function.

brianrho, 2019-02-19:
If reading templates now works, I'll be closing this. I've added a warning to the readme regarding the R551 and its problems. Could still be that the command is supported but with a different code but you'll have to contact the manufacturer for that or experiment yourself.

vvvait, 2019-02-20:
ok, i found few commands with answer
1A -> success
47 -> looks like verison PPMPDG62S200V100
53 -> error x55
68 -> x00 x01

brianrho, 2019-02-20:
Great. If you get more info about the DOWNCHAR, let me know

closed 2019-02-20 by brianrho