brianrho/FPM #19: R302/R551 Did not find fingerprint sensor

minswho, 2019-04-13:
> [+]Response timeout
> 
> Did not find fingerprint sensor :(

I tried switching RX/TX and Adafruit example even. (pin 2 and 3, Arduino Micro)
Based on closed issue, I tried using set_password example as well.

Nothing seems to work. If anyone can suggest some troubleshooting suggestions, I'd really appreciate it.

brianrho, 2019-04-13:
What is the name of your sensor? Have any of the examples worked so far? Which ones? What was the last example that worked?

minswho, 2019-04-14:
R302 and R551 ^. I tried set_password,enroll, and Adafruit example (set_password and enroll) ^, none of them worked.

brianrho, 2019-04-14:
Which did you try first?

minswho, 2019-04-14:
Enroll example from this repo

brianrho, 2019-04-14:
Its likely you've changed the password already, especially with using `set_password` from 2 different repos. You'll have to try using `finger.begin(0x12345678)` in the `enroll` example and see if that gets you access.

minswho, 2019-04-14:
I'll try that, but it never detected the fingerprint sensor once. Should I try to somehow reset the device?

And with R551, besides uncommenting that one line, do I need to do anything else to make it work? (looking for basic register & match fingerprint features)

brianrho, 2019-04-14:
I'm surprised it never detected the sensor. The first step is always the `begin()` call, you must get the `found fingerprint sensor` message before you can continue. If you're certain `FPM` is the first library you used after you purchased the sensor and yet it has never been detected with this or Adafruit's, then you likely have wrong wiring. So you should check that RX/TX are connected correctly and you're providing VCC/GND at the right pins.

minswho, 2019-04-14:
I don't exactly remember the first library I tried tbh, but I followed the
data sheet on the wiring (and even tried switching rx/tx).

I'd be happy to send over pictures, but I don't think that's it.

Is there a way to reset the device? And what example were we supposed to
use first? (set_password?)

brianrho, 2019-04-14:
No, there's a warning at the top of `set_password` that it will change your password, there's usually no need to run it unless you know what you're doing. `enroll` is the first example you should usually run to add new prints. It's important to remember the first library you tried and what results you got because it helps determine exactly what settings could've been changed and what they are now. 
Send a link to the product (powered with 5V, right?). And also try `finger.begin(1337)` in case the Adafruit example changed the password first.

brianrho, 2019-04-14:
There's no way to reset it, that I know of. If you've changed the password, you just have to try and figure out the new password.

minswho, 2019-04-15:
Tried both password with R302, same problem unfortunately.

R551 has a chip reset pin, so I tried resetting and tried both password in enroll, and it's still returning the "did not find" error

I attached wiring pictures for both sensors (R551 takes 3.3V)
https://imgur.com/a/jgNlNAy

I don't know man

brianrho, 2019-04-16:
You can try a different 3.3V source (like the one on a USB-UART adapter) in case its a power supply issue, I remember most Arduinos usually can't source more than 50 mA from their 3.3V pin. Also use a multimeter to verify that all 4 connections were really made, breadboards are often unreliable.
Not much else that can be done if the sensors don't even respond and have never been detected.

closed 2019-06-18 by brianrho