brianrho/FPM #20: Verify Sensor

oskarn97, 2019-06-11:
Hello!
How do I make sure that nobody exchanges the sensor attached to my controller board?
Can I change the address of the sensor?

brianrho, 2019-06-11:
You can change the address of the sensor even if its not implemented yet. But that wouldn't prevent someone from replacing your sensor. Every command is preceded by the address and password, an "attacker" only needs to spy on the serial lines to learn both of them and change their own sensor settings to match. If there was some builtin unique read-only ID for each sensor, you could use that but as far as I know, there isn't one for these sensors. 
What exactly are you trying to protect against?

oskarn97, 2019-06-11:
Thanks for that information!
I only want to protect against physically exchanging the sensor (without sniffing).
Even if I do not have the correct password, a sensor will still report it detected a valid Fingerprint if exchanged.

brianrho, 2019-06-12:
Okay, I get the first part, you're not worried about sniffing but I'm not sure about the rest? If the wrong password (or address) is used, the sensor will simply refuse to respond to further commands, valid fingerprint or not

oskarn97, 2019-06-12:
In my testing with the R551, although using a wrong password, the sensor still reported a valid fingerprint which caused my controller to accept it. I've changed the address now, that works and the exchanged sensor no longer accepts commands from the controller. I've created a pull request with the added functions.

closed 2019-06-12 by oskarn97