brianrho/FPM #22: Send image via UDP

hell77, 2019-07-04:
Hello

Is there a way to send image not to the serial port but via UDP?

Thank you in advance

hell77, 2019-07-04:
If I use the standard library EthernetUdp, I get an error

```
[+]Got header
[+]Address: 0xFFFFFFFF
[+]PID: 0x2
[+]Length: 128
7A DD 97 65 43 34 67 AC DE EC B8 64 34 57 9B CC DD C9 74 21 12 34 57 9A DD EE DD B6 44 46 9C CC B9 83 22 23 45 7A CD CB 97 68 75 44 57 AB BA 86 43 22 22 24 56 79 43 37 57 77 88 AD 9C EE EE FF AC 12 83 86 87 64 CC 84 66 86 EE DD FF 35 59 ED 22 22 35 78 EE BB FF 76 DB EE 4C 82 8A AA 95 22 DE 78 AA BA CC 95 9A AD EE EF CC DE A8 65 78 BA DD ED BB 75 BD 42 EE DD FF DD 21 46 CE BC 32 DD 
[+]Wrong chksum: 0x5566

[+]Got header
[+]Address: 0xFFFFFFFF
[+]PID: 0x2
[+]Length: 128
A6 43 32 47 CD DD DE ED B8 53 11 23 6A BD EE ED DD A5 21 11 14 7A BC DD B8 53 23 46 9C DD ED A7 42 11 23 69 CD CB CC EE EB 96 43 22 23 34 44 45 77 67 78 AA 78 BE ED 97 56 79 99 86 89 47 AD 8A 9B EE 42 A5 B6 11 AD 69 45 8A 55 79 7A 43 DE DD DD FF 8A 46 68 B8 ED AA A9 DD 89 22 DE 45 EE DD 2 67 CB CA 34 B9 9C CC B5 55 68 33 CD EE EF AC EE CB 35 22 64 9B DC 88 DC 34 96 DD DC FF DC 22 
[+]Wrong chksum: 0x76BC
```

brianrho, 2019-07-04:
Yeah, you can just write to your `Client` object:

`finger.readRaw(FPM_OUTPUT_TO_STREAM, &client, &read_finished);`

The function takes a `Stream` object.

brianrho, 2019-07-04:
Debug level 2 prints everything, which slows stuff down and you'll miss image bytes from the sensor. For a high-speed lengthy stream like this, best set it to 1 so you only get the error messages. 
Where's the code that produced this?

hell77, 2019-07-04:
Thanks for your reply

With debug level 1

```
Image taken
Starting image stream...
	
[+]Wrong chksum: 0x74DC
[+]Wrong chksum: 0xFFDD
[+]Wrong chksum: 0xDD
[+]Wrong chksum: 0x46
.....
[+]Wrong chksum: 0xBBBC
[+]Wrong chksum: 0xBBBC
[+]Response timeout
[+]Wrong read length: -1
Error receiving packet 0
```

[testfp.zip](https://github.com/brianrho/FPM/files/3359533/testfp.zip)

brianrho, 2019-07-04:
Use pastebin.com for the code

brianrho, 2019-07-04:
Also provide the name of your sensor and microcontroller

hell77, 2019-07-04:
https://pastebin.com/jjGSAA9k
Arduino UNO rev3
sensor don't now exactly (2 minutes I post a fotos)

hell77, 2019-07-04:
[image]
[image]

brianrho, 2019-07-04:
Hmm, haven't seen this before. Does streaming images to your PC work?

hell77, 2019-07-04:
I think - yes.

In port monitor I see 

```
Waiting for a finger...
Image taken
Starting image stream...
	⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮ݤ5⸮̻⸮T3Zݓkݦ23I⸮⸮TDDY⸮⸮R6⸮⸮⸮r͂7⸮⸮!⸮ޕ"$~⸮B^⸮"]⸮"~⸮b#j⸮B#9ޕe⸮⸮⸮TD⸮⸮⸮⸮⸮⸮⸮
…
36864 bytes read.
Image stream complete.
```
Also I has your oldest library and it work very well with sensor. But only via serial port

brianrho, 2019-07-04:
Which version are you using right now, that yields this output? The latest version or an older one?

hell77, 2019-07-04:
Now latest

brianrho, 2019-07-04:
I think the UDP write is just taking too long to buffer the data and you're losing bytes from the sensor.

Try commenting out this line at 734: `outStream->write(byte);` in `FPM.cpp` and see if it succeeds this time. Don't change anything else, keep the debug level at 1.

hell77, 2019-07-04:
You are right, there are no errors.

I tried to change my code like this

```
while (true) {
        Udp.beginPacket(rip, rPort);
        Udp.write(fserial.read());
        Udp.endPacket();
        
       /*bool ret = finger.readRaw(FPM_OUTPUT_TO_STREAM, &Udp, &read_finished);
        if (ret) {
            count++;
            if (read_finished)
                break;
        }
        else {
            Serial.print("\r\nError receiving packet ");
            Serial.println(count);
            return;
        }
        yield();*/
    }
```

And I get everything on the remote computer, each byte separately.

brianrho, 2019-07-04:
How many bytes did you receive exactly? Still possible, even likely, that you didn't get everything

hell77, 2019-07-04:
I think that if add before and after the 734 line

```
Udp.beginPacket (rip, rPort);
outStream-> write (byte);
Udp.endPacket ();

```
But I do not know how to do it.

hell77, 2019-07-04:
> How many bytes did you receive exactly? Still possible, even likely, that you didn't get everything
I did this for the test.
There is an endless cycle.

```
  while (true) {
       Udp.beginPacket (rip, rPort);
         Udp.write (fserial.read ());
         Udp.endPacket ();
}
```
Therefore, I can not count how many bytes

Maybe my questions are stupid, I know C ++ very bad.

brianrho, 2019-07-04:
No, I mean, you should count the bytes you received at your remote server. It should be at least 40kB.

hell77, 2019-07-04:
If I understand correctly, you are not writing to the stream all bytes, only content.  

I do not understand how to do this and how to detect the end of data.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

brianrho, 2019-07-04:
Yes, I only stream the contents, but leave that for now. Do you have any code on your server to receive the data? How do you know the data even arrived at all?

hell77, 2019-07-04:
yes, I have.  I display everything on the screen

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

brianrho, 2019-07-04:
Then maybe your code can count what it receives. 

If you can't count what you received, then you can try this instead: 
Undo all your changes so far, go back to the original sketch you sent me. Then in your Arduino installation folder, search for `SoftwareSerial.h` and change the value of `_SS_MAX_RX_BUFF` from 64 to 255

hell77, 2019-07-04:
Good.  I'll do everything in the morning, it's already night here.  I can also count incoming bytes. Thanks

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

hell77, 2019-07-05:
Hello
I changed the value to 255 and got other errors

```
[+] Packet too long: 61150
[+] Response timeout
[+] Wrong read length: -1
Error receiving packet 2
```

brianrho, 2019-07-05:
Declare this at file scope.

```
#define BUFSZ    128
`uint8_t buffer[BUFSZ];`
```

Then change your existing code like this:
```

    bool read_finished;
    int16_t count = 0;
    uint16_t readlen = BUFSZ;
    Udp.beginPacket(rip, rPort); // delete it from loop()

    while (true) {
        bool ret = finger.readRaw(FPM_OUTPUT_TO_BUFFER, buffer, &read_finished, &readlen);
        if (ret) {
            count++;
            Udp.write(buffer, readlen);
            Udp.endPacket();   // delete it from loop()

            readlen = BUFSZ;
            if (read_finished)
                break;
        }
        else {
            Serial.print("\r\nError receiving packet ");
            Serial.println(count);
            return;
        }
        yield();
    }
```

It will probably still raise errors but send everything you get this time.

hell77, 2019-07-05:
I received exactly 128 bytes, 2 times in a row (the same array).
And every time I get 2 times.

brianrho, 2019-07-05:
Yeah, send the errors, send everything that was printed

hell77, 2019-07-05:
**Arduino** 

```
Waiting for a finger...
Image taken
Starting image stream...
	
[+]Wrong chksum: 0xDDDB
[+]Response timeout

[+]Wrong read length: -1

Error receiving packet 2
```

**Remote computer received**

```
<-- 05.07.2019 12:22:50(750) | Client: 192.168.3.177:8888 | Length: 128: 
137
168
188
205
216
103
138
237
219
169
149
51
55
189
221
221
150
121
151
156
221
221
219
136
156
205
221
221
220
150
36
190
221
221
218
84
139
187
204
221
221
221
151
120
155
205
221
221
220
204
200
84
71
205
221
221
217
153
155
221
221
236
98
55
237
234
51
59
237
221
220
116
34
108
220
217
158
221
219
101
89
221
222
222
134
86
121
205
221
221
218
152
68
205
221
221
168
171
153
237
221
221
204
202
188
221
221
220
186
153
237
221
221
220
221
221
220
122
221
221
221
221
221
221
221
221
221
221

<-- 05.07.2019 12:22:50(811) | Client: 192.168.3.177:8888 | Length: 128: 
137
168
188
205
216
103
138
237
219
169
149
51
55
189
221
221
150
121
151
156
221
221
219
136
156
205
221
221
220
150
36
190
221
221
218
84
139
187
204
221
221
221
151
120
155
205
221
221
220
204
200
84
71
205
221
221
217
153
155
221
221
236
98
55
237
234
51
59
237
221
220
116
34
108
220
217
158
221
219
101
89
221
222
222
134
86
121
205
221
221
218
152
68
205
221
221
168
171
153
237
221
221
204
202
188
221
221
220
186
153
237
221
221
220
221
221
220
122
221
221
221
221
221
221
221
221
221
221
```

brianrho, 2019-07-05:
If something's especially lengthy, please use pastebin.

Now try to change the baud rate to 9600 with the `set_parameters` sketch. Let's see if slowing down the serial port will be enough for the UDP to keep up.

hell77, 2019-07-05:
> If something's especially lengthy, please use pastebin.

Ok, sorry

```
SET PARAM test
Found fingerprint sensor!
Capacity: 256
Packet length: 128
Send any character to change the module baud rate...
Unknown error 105
```

brianrho, 2019-07-05:
Okay, now try your image transfer sketch again. Just change the `SoftwareSerial` baud rate to 9600.

brianrho, 2019-07-05:
If the sensor doesn't get detected with 9600, change it back to 57600. It's likely the command didn't succeed since you got this error.

brianrho, 2019-07-05:
Also send a link to this product, where you bought it.

hell77, 2019-07-05:
Sorry for not answering for a long time, I didn’t want to disturb you again. Experimenting a bit.

Oк.
You were absolutely right. It was necessary to reduce the speed of the serial port.

Unfortunately, I could not find how to do it for my sensor, so I had to buy another one (model DY50) :). I do not know where they bought the old one.

Experimentally, I’ve found out that the 19200 Baudrate is working.

The image is created successfully.
True, I did not understand why need to add each byte 2 times when creating an image.

On Monday, I will finish all the files and send them to you. I hope they will be useful.

brianrho, 2019-07-05:
Okay, that's fine, DY50 is one of the more common and trusted sensors. 
So you mean 19200 works over Serial? Or is it over UDP?

Duplicating each byte is one way to extrapolate and get back the (roughly) original 256x288 image. The image that's sent from the sensor is already somewhat compressed, in that only the high nibble of each pixel is transmitted. So we're basically just guessing about what was in the lost low nibbles.

hell77, 2019-07-05:
with duplication I understood. Thanks.

I mean over UDP, but only in FPM_OUTPUT_TO_BUFFER mode

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

brianrho, 2019-07-05:
Okay, send the files when you can. I'll be closing the issue after that

hell77, 2019-07-08:
Hello
In the attachment is an example of use. 
For Arduino and Windows (C#).

[SendToUDP.zip](https://github.com/brianrho/FPM/files/3367440/SendToUDP.zip)

Thank you very much for the help.

brianrho, 2019-07-08:
You're welcome. The sample sketch in the zip just sends a string. Could you send a paste of the image transfer sketch, just so I can be sure its what I expect?

hell77, 2019-07-08:
I am so sorry.

Many files open at the same time.
Attached another file was wrong.

[SendToUdp.zip](https://github.com/brianrho/FPM/files/3368288/SendToUdp.zip)

Must be right now

brianrho, 2019-07-08:
Sure, it's fine now

closed 2019-07-08 by brianrho