brianrho/FPM #23: Fast Search?

nicoruy, 2019-08-13:
Hi,
I'm trying to use your fork of the adafruit library, migrating from the original one.
My goal is to download and upload a set of fingerprints based on a menu selection. I´m just starting with tthat I´m not sure if it´s even possible.
 
During my migration I noticed that most of the functionality is the same or extended (which is great!) but I found that FastSearch is not available.
The packet id FPM_HISPEEDSEARCH is there but never used.
Is there any particular reason why this is not available on the fork?
Which is the intended method to check if a fingerprint matches?

Thanks !

brianrho, 2019-08-13:
I replaced `HISPEEDSEARCH (0x1B)` with `SEARCH (0x04)` because it seems some devices (like the R551) don't support the former, or time out when they receive it. I would also say I've found no measurable difference in speed between the two commands, but then I've not tried them with hundreds of prints. If you have enough (unique) prints, you could test both (just replace the hex number in the header) and let me know how they perform.

brianrho, 2019-08-15:
If you've no other questions, I'll be closing this.

closed 2019-08-15 by brianrho

nicoruy, 2019-08-22:
My sensor has 143 registered fingerprints (and a limit of 150).
I used your suggestion and tried both SEARCH and HISPEEDSEARCH with fingerprints registered on the firsts positions and last positions of the sensor.
Didn´t find any noticeable difference between both methods, so I think it´s ok to use SEARCH

Thanks

brianrho, 2019-08-23:
Good to know, thanks!