brianrho/FPM #24: [+]Response timeout

TehseenHasan, 2019-08-14:
I am using Wemos D1 Mini, ESP8266-12E based development board. ESP8266 is controlling R301T Fingerprint sensor using Serial Monitor commands. I am also using ESP8266's WiFi as Access Point to do some configurations and communications with local clients. Everything is working as expected BUT only one problem is there...!!!

The Serial Monitor showing this error randomly even when WiFi is disabled (No AP mode).
`[+]Response timeout`

Some times, when AP mode is enabled, the error looks like this:
```
[+]Wrong chksum: 0xC
[+]Response timeout
```
or this:
```
[+]Wrong chksum: 0xB
[+]Response timeout
```
This timeout error becomes more frequent when ESP8266 is working in Access Point Mode and communicating with Clients.

Does this timeout occurs due to some **Timer issues**???
I can not understand the reason of this error and how to solve it. Please help me to solve this issue. Thanks!

brianrho, 2019-08-14:
What command gives this error? What's your baud rate? Post a minimal example of the code that yields this behaviour (with Pastebin.com)

TehseenHasan, 2019-08-16:
When Client connected to ESP8266 and it makes some requests like **call a URL** then timeout error appears in Serial monitor...

**This is the minimal code:** https://pastebin.com/0S5Q7vNi

**Sometimes it also give this error:** 
`[+]Packet too long: 128`

**While simple timeout error is very frequent:**
[image]

brianrho, 2019-08-16:
First, fix the return type of `get_template_count` and `rtn_val` to `int16_t` if you'll be using negative values. 
Are you using the most recent version of the library and is the current debug level ERRORS ONLY? Check `FPM.h` to be sure.
Does this timeout error occur only when you try to get the template count? Try the template count example a few times and see if you still get any error.

TehseenHasan, 2019-08-17:
> First, fix the return type of `get_template_count` and `rtn_val` to `int16_t` if you'll be using negative values.

Okay I changed the data type.

> Are you using the most recent version of the library and is the current debug level ERRORS ONLY? Check `FPM.h` to be sure.

Yes I'm using most recent version. I just downloaded the library from here at github. And Debug level is default (Errors Only) in FPM.h

> Does this timeout error occur only when you try to get the template count? Try the template count example a few times and see if you still get any error.

No, timeout error occurred no matters I count the fingerprints, Enroll a new Fingerprint or Search. Timeout occurs each time randomly (mostly when WiFi is ON).

**Important:**
Timeout occurs very rarely when WiFi is OFF and I just give commands through Serial. But as soon as I Turn WiFi ON and Client is connected to ESP8266 timeout error start coming more and more...

See this is the latest results:

[image]

brianrho, 2019-08-17:
You're powering the circuit through your Wemos USB port, right? Do you have anything else connected to the Wemos besides the R301T? Since the timeouts only happen when you turn on the Wi-Fi (soft AP?), it's possible that you're simply not providing enough power for the ESP8266 and R301T. So try powering the R301 with some other source (maybe another USB port).

Also try increasing `FPM_DEFAULT_TIMEOUT` to 5000 and see if it still times out.

TehseenHasan, 2019-08-18:
**1.** Yes I was powering Wemos with its MicroUSB port. There is a Buzzer Module and an LED in my circuit along with Wemos D1 Mini and Fingerprint Sensor. 
Now I attached **5V 2A** external power supply to this circuit. But **error did not resolve**. Timeout still occurs .

**2.** I changes `FPM_DEFAULT_TIMEOUT` value to 5000, 10000 and 500 too but **timeout error is still there**...  

I noted that when Timeout error occurs **fingerprint module stop searching** (stuck) for **FPM_DEFAULT_TIMEOUT** seconds. I mean after error appear **fingerprint sensor stops working** and then **start working again after the timeout value** seconds set for `FPM_DEFAULT_TIMEOUT`. 
So to avoid the bad behavior of stop responding of the fingerprint sensor during timeout error I just changes the `FPM_DEFAULT_TIMEOUT` value **100** hence when timeout error occurs it stops sensor's working only for 0.1 seconds and then sensor is alive again. So by doing this, somehow now my system is working despite of **timeout errors are still there** but at least my fingerprint sensor does not stops and stuck.

I think there are some sort of **compatibility issues** of this FPM Library with ESP8266 Modules.
I'm not an expert but please consider this issue and update the library, if possible, to add compatibility for ESP8266 modules. Thanks!

TehseenHasan, 2019-08-18:
**1.** Yes I was powering Wemos with its MicroUSB port. There is a Buzzer Module and an LED in my circuit along with Wemos D1 Mini and Fingerprint Sensor. 
Now I attached **5V 2A** external power supply to this circuit. But **error did not resolve**. Timeout still occurs .

**2.** I changes `FPM_DEFAULT_TIMEOUT` value to 5000, 10000 and 500 too but **timeout error is still there**...  

I noted that when Timeout error occurs **fingerprint module stop searching** (stuck) for **FPM_DEFAULT_TIMEOUT** seconds. I mean after error appear **fingerprint sensor stops working** and then **start working again after the timeout value** seconds set for `FPM_DEFAULT_TIMEOUT`. 
So to avoid the bad behavior of stop responding of the fingerprint sensor during timeout error I just changed the `FPM_DEFAULT_TIMEOUT` value to **100** hence when timeout error occurs it stops sensor's working only for 0.1 seconds and then sensor is alive again. So by doing this, somehow now my system is working despite of **timeout errors are still there** but at least my fingerprint sensor does not stops and stuck.

I think there are some sort of **compatibility issues** of this FPM Library with ESP8266 Modules.
I'm not an expert but please consider this issue and update the library, if possible, to add compatibility for ESP8266 modules. Thanks!

TehseenHasan, 2019-08-18:
**1.** Yes I was powering Wemos with its MicroUSB port. There is a Buzzer Module and an LED in my circuit along with Wemos D1 Mini and Fingerprint Sensor. 
Now I attached **5V 2A** external power supply to this circuit. But **error did not resolve**. Timeout still occurs .

**2.** I changes `FPM_DEFAULT_TIMEOUT` value to 5000, 10000 and 500 too but **timeout error is still there**...  

I noted that when Timeout error occurs **fingerprint module stop searching** (stuck) for **FPM_DEFAULT_TIMEOUT** seconds. I mean after error appear **fingerprint sensor stops working** and then **start working again after the timeout value** seconds set for `FPM_DEFAULT_TIMEOUT`. 
So to avoid the bad behavior of stop responding of the fingerprint sensor during timeout error I just changed the `FPM_DEFAULT_TIMEOUT` value to **500** hence when timeout error occurs it stops sensor's working only for 0.5 seconds and then sensor is alive again. So by doing this, somehow now my system is working despite of **timeout errors are still there** but at least my fingerprint sensor does not stops and stuck.

I think there are some sort of **compatibility issues** of this FPM Library with ESP8266 Modules.
I'm not an expert but please consider this issue and update the library, if possible, to add compatibility for ESP8266 modules. Thanks!

brianrho, 2019-08-19:
Can't add compatibility if the problem's cause is unknown, and that's assuming the issue is even with the library. This is the first time I've seen an R301 in use, so there's no precedence here. I'd recommend you try the Adafruit library and see if it works for you without timeouts. Then we could conclude that this library is likely the problem.

TehseenHasan, 2019-08-19:
Okay

TehseenHasan, 2019-08-19:
Okay but I see that this timeout error is discussed by someone else too. 
https://github.com/brianrho/FPM/issues/18#issuecomment-471733306

brianrho, 2019-08-19:
That was for a different MCU and sensor. And in that case, it turned out the high speed search command wasn't supported by the module and so it was replaced with the regular search command, as well as increasing the timeout to a more reasonable 2 seconds. Yours is more general and seems more like a case of poor power supply or the module simply doesn't support these commands as documented. If you have a different sensor model, you can try it to see if the problem goes away. Or try a different library, like I already suggested.

TehseenHasan, 2019-08-20:
Okay...

closed 2019-08-20 by TehseenHasan