brianrho/FPM #28: Image to pc

kaushalsapre, 2019-09-23:
Hey,

I'm trying to extract a fingerprint image to my computer using the image to pc arduino + python codes. 

I'm running an R307 fingerprint sensor with an Arduino Uno connected to an Ubuntu 18.04 with Python 3.6

Everything seems to be okay on the arduino side. I can get to a stage where the 'Image stream complete' message shows up. I use the following values for the serial port settings in the python code - 
Arduino serial port number - /dev/ttyACM0
baud rate - 57600
I've triple checked the above values. When I enter the file name for the image to be stored, the code gives me the error - Invalid port settings! Image extraction failed!

After which it produces a glitched image on my designated folder. 

Wondering where I am going wrong with this? When I try changing the baud rate from 57600 to anything else, the code does not recognize the sensor. 
[image]

brianrho, 2019-09-23:
The python script's having trouble opening the port for some reason. Either you've gotten the port settings wrong or some other device is already using the port, like your Serial Monitor. For the former, I assume you've thoroughly checked the port settings. For the latter, make sure all serial windows are closed before you run the script. 
If it still doesn't work, I've made a small change to the script so that it prints the exact exception. So download the latest version, try again, and send a screenshot of what you get.

brianrho, 2019-10-16:
I assume one of those solved your problem.

closed 2019-10-16 by brianrho