brianrho/FPM #29: Best sensor to use?

nilsmelchert, 2020-02-27:
Hi,
At first, thank you for providing this code. I have a couple of questions regarding this library.

- Is there a sensor which worked best in the past? 

- Is it also possible to use capacitive sensors?

- Did I get that right from the documentation, that I am also able to read out the "raw" fingerprint from the sensor to the computer?

Thanks a lot for your answers

nilsmelchert, 2020-02-27:
Sorry. I should have read more before asking questions. ;-)

closed 2020-02-27 by nilsmelchert

brianrho, 2020-02-28:
Glad you found it useful :-)
As for reliable sensors, most of them out there are decent, the only limitation they have individually is their supported features and I've recommended a few in the README that offer most things people need. But you can verify with their datasheets to be sure they support the features you want.