brianrho/FPM #32: R503 led_control problem

ardiwidiyono, 2020-06-09:
hi please help me R503 sensor led_control problem "no matching function for call to 'FPM :: led_control ()' '

brianrho, 2020-06-11:
Well, you can clearly see the function prototype requires 4 arguments so why are you calling it without any? I don't have any examples at the moment. If you can find an R503 datasheet, I may be able to add an example to the library.

ardiwidiyono, 2020-06-15:
Is this the R503 data sheet that you need ??

[R503 fingerprint module user manual.pdf](https://github.com/brianrho/FPM/files/4780895/R503.fingerprint.module.user.manual.pdf)

brianrho, 2020-06-16:
Yes, this seems to have some documentation on the command but it's not very clear. I'll look into it and maybe have an example out in a few days. Since I don't have the hardware, you'll have to do all the testing on your end.

ardiwidiyono, 2020-06-16:
thank you very much thank you ...
I patiently waited .... 
do you have a youtube channel ??

brianrho, 2020-06-16:
Nah, I don't have a platform like that. 

To be clear, can you still use the sensor, without controlling the light? Right now, it's controlled automatically when you place your finger, right?

ardiwidiyono, 2020-06-17:
if I use fingerprint enrollment, it can be saved, but if I use examples the fingerprint sensor is not readable ... 
when I use enroll LED it doesn't show any response ...

brianrho, 2020-06-18:
How do you mean the sensor is not detected? Aren't you using it for enrollment, as you said? Specifically, which exact examples do not work?

ardiwidiyono, 2020-06-28:
the sensor is running normally, yesterday I made a mistake in determining the rx tx .... 
please help me with the R503 led control problem

Osrx, 2020-10-22:
> the sensor is running normally, yesterday I made a mistake in determining the rx tx ....
> please help me with the R503 led control problem

finger.led_control(1, 30, 1, 15);

LED Works like a charm

closed 2023-12-26 by brianrho