brianrho/FPM #33: Get free/available index for new fingerprint enrollment

TehseenHasan, 2020-06-24:
I am enrolling and deleting many fingerprints one after another on random indexes for testing purposes. 
I want to find the very first available index for next enrollment. I read the FPM.cpp file and find this function.
`int16_t FPM::getFreeIndex(uint8_t page, int16_t * id)`
Is this function is for this task? and is there any example code available to use this function?
Thanks.

TehseenHasan, 2020-06-24:
I am using this code in Wemos D1 Mini with R300 Fingerprint Sensor Module.
And this function `getFreeIndex` worked for me. It is a clean way to do this task. It saved a lot of my time.
Thanks.

```
#include <SoftwareSerial.h> // espSoftwareSerial Library V6.8.3
#include <FPM.h> 

//Fingerprint Sensor Initilizations
SoftwareSerial fingerprint(D6, D7); // RX, TX
FPM finger(&fingerprint);
FPM_System_Params params;

void setup()
{
  Serial.begin(9600);
  fingerprint.begin(57600);

  if (finger.begin())
  {
    finger.readParams(&params);
    Serial.println("Found fingerprint sensor!");
    Serial.print("Capacity: "); Serial.println(params.capacity);
    Serial.print("Packet length: "); Serial.println(FPM::packet_lengths[params.packet_len]);
    Serial.println();
  }
  else
  {
    Serial.println("Did not find fingerprint sensor!!! \n");
    while (1) yield();
  }

  get_free_index();

}

//Function to find the very first available index in the Fingerprint Sensor for new fingerprint enrollment
int16_t get_free_index()
{
  int16_t index = 0;
  uint8_t page = 0;
  int16_t x = finger.getFreeIndex(page, &index);

  Serial.print("Free index is: ");
  Serial.println(index);
  return index;
}
```

closed 2020-06-24 by TehseenHasan

brianrho, 2020-06-24:
You're welcome :-) FYI, the `enroll` example shows how to use that method, to scan _all_ the flash pages, not just the first page as you're presently doing:

`bool get_free_id(int16_t * fid) {
    int16_t p = -1;
    for (int page = 0; page < (params.capacity / FPM_TEMPLATES_PER_PAGE) + 1; page++) {
        p = finger.getFreeIndex(page, fid);
        switch (p) {
            case FPM_OK:
                if (*fid != FPM_NOFREEINDEX) {
                    Serial.print("Free slot at ID ");
                    Serial.println(*fid);
                    return true;
                }
                break;
            case FPM_PACKETRECIEVEERR:
                Serial.println("Communication error!");
                return false;
            case FPM_TIMEOUT:
                Serial.println("Timeout!");
                return false;
            case FPM_READ_ERROR:
                Serial.println("Got wrong PID or length!");
                return false;
            default:
                Serial.println("Unknown error!");
                return false;
        }
        yield();
    }
    
    Serial.println("No free slots!");
    return false;
}`