brianrho/FPM #34: templates example fails

cheofusi, 2020-07-20:
I'm unable to use the read_template function in the templates example with the FPM10A sensor. Fails with a "packet too long: 130" error.

brianrho, 2020-07-21:
Hi Brandon, good work narrowing it down:

https://github.com/brianrho/FPM/blob/7b2c0cf8fa15d52d431888bec2a0aeecee74ec7b/src/FPM.cpp#L721

From what you explained earlier, this condition is what is failing and yet your packet length (130 bytes) looks just fine to me (usually 128 + 2 bytes for checksum, as I recall). So, I think the problem here is, you supplied an insufficient buffer to `readRaw()`. If you're using the default `templates` example, the default buffer size there is 768. If you check the comment above the relevant #define, you'll see where I mentioned that some sensors have larger-than-normal template sizes, up to 1024 bytes. It's possible your sensor is one of those, so try setting `BUFF_SZ` to 1024 and see if it helps.

brianrho, 2020-07-24:
Wait, that doesn't make sense. I thought you said the error was "packet too long: 130". Are you saying it reports 130 as the length **within the packet itself** (not the initial system parameters) but then follows up with 267 bytes instead?

brianrho, 2020-07-24:
Because the only time I use `sys_params.packet_len` is when writing/uploading templates *back* to the sensor.

cheofusi, 2020-07-25:
> Wait, that doesn't make sense. I thought you said the error was "packet too long: 130". Are you saying it reports 130 as the length **within the packet itself** (not the initial system parameters) but then follows up with 267 bytes instead?

Exactly. I can't figure out whether it's the ACK packet following the download command or its the first packet containing the template itself. Same issue with uploading templates to the sensor. Have you tested the _WriteRaw_ and _ReadRaw_ functions with the fpm10a??

brianrho, 2020-07-25:
Definitely. An FPM10A was actually the first fingerprint sensor I ever used years back. You likely just have different/updated firmware. 

Leave template uploads for now. Let's focus on just downloads. 

- When you make BUFF_SZ = 1024 for the total buffer size, do you still get this exact same error? Provide the entire serial logs, as well as your sketch using pastebin.com.
- I've just checked the Adafruit example and the implementation isn't reliable at all, can't be used here as a benchmark. The "267 bytes" thing is just a comment based on the likely wrong assumption that each packet has 256 bytes of content. All it does is just read everything it can, into a buffer, at most 534 bytes, regardless of the actual size of the template, no parsing at all, except to strip away the packet header and checksum. 

I still think you just have a buffer that's too small. So try the 1024-sized buffer, as I said before, and let me know what you find.

cheofusi, 2020-07-27:
I tried all the 1024. The sketch is the templates example. By the way not every packet is 267 bytes in length. Only the data packets carrying the template. I also solved the problem for uploads. All I did was change the chunk_sz in the readRaw function to 256 and the loop counter, i, in the writePacket function to an unint16_t.

brianrho, 2020-07-27:
Okay, if your issue is solved, I'll be closing this.

closed 2020-07-27 by brianrho