brianrho/FPM #37: issue with sunlight

irpcpro, 2020-11-02:
hi . i use ZFM60 for my project and i use from your library for this module.
everything is Ok but i have a issue ...
when module is under the sunlight , the module want to try read finger and return error .. without put finger on it .
it means the Ghost try to put his finger :))) but system cannot recognized and return error (not found in database) .
but into building everything is ok .

how can i fix the sunlight issue ?
is this for security level ?
how can i change the security level on this line of library ?
is this line for change security ?
src/FPM.h:134
`/* security levels */
enum {
    FPM_FRR_1 = 1,
    FPM_FRR_2,
    FPM_FRR_3,
    FPM_FRR_4,
    FPM_FRR_5
};`

brianrho, 2020-11-04:
In order to know if security levels is even relevant to your issue, you need to first provide your sketch and also the Serial Monitor output. I think you're likely not using the getImage()/Image2Tz() API properly.

irpcpro, 2020-11-05:
its true, all code works good on the building for days ....
i try again and it seems this part :
p = finger.getImage();
detect the sunlight as a fingerprint shadow ...

how can i change security ZFM60 by set parameter ????
on the examples just is setting for changing baud rate ..

brianrho, 2020-11-10:
> you need to first provide your sketch and also the Serial Monitor output

closed 2023-12-26 by brianrho