brianrho/FPM #38: Upload template to AS608

farhad3113, 2020-12-04:
I have a problem with uploading the downloaded template from AS608 FPM (also known as JM-101) and get "Packet too long: 130" in serial.
I also changed BUFF_SZ to 1024 so I able to download the template from the sensor.

brianrho, 2020-12-05:
Are you using the default example? No change at all except to change the BUFF_SZ?
Can you post your entire serial output/log?

farhad3113, 2020-12-05:
No, it's not the default example.
I found out that templates move successfully to a new place and I can access them in that new fid but it returns "packet too long:130"

**my functions:**

```
bool uploadTemplate(uint16_t fid, uint8_t * buff, uint16_t buff_sz){
    #ifdef DEBUG_MODE
      debugSerial.println("[debug] ==== Upload FingerPrint ====");
    #endif
    uint16_t p = -1;
    p = finger.uploadModel(fid);
    if(p != FPM_OK){
        #ifdef DEBUG_MODE
          debugSerial.println("[debug] FPM: Error in Upload!");
        #endif
        return false;
    } 
    yield();
    finger.writeRaw(buff, buff_sz);
    
    p = finger.storeModel(fid);
    if(p != FPM_OK){
        #ifdef DEBUG_MODE
          debugSerial.println("[debug] FPM: Error in Store!");
        #endif
        return false;
    } 
    return true;
}

uint16_t donwloadTemplate(uint16_t fid, uint8_t *buff, uint16_t buff_sz){
    #ifdef DEBUG_MODE
      debugSerial.println("[debug] ==== Download FingerPrint ====");
    #endif
    uint16_t p = -1;
    p = finger.loadModel(fid);
    if(p != FPM_OK){
        #ifdef DEBUG_MODE
          debugSerial.println("[debug] FPM: Error in Load Model!");
        #endif
        return 0;
    }
    p = finger.downloadModel();
    if(p != FPM_OK){
        #ifdef DEBUG_MODE
          debugSerial.println("[debug] FPM: Error in Download!");
        #endif
        return 0;
    }
    bool read_finished;
    int16_t count = 0;
    uint16_t readlen = buff_sz;
    uint16_t pos = 0;
    while (true) {
        bool ret = finger.readRaw(FPM_OUTPUT_TO_BUFFER, buff+pos, &read_finished, &readlen);
        if (ret) {
            count++;
            pos += readlen;
            readlen = buff_sz - pos;
            if (read_finished) { break; }
        }
        else {
            #ifdef DEBUG_MODE
              debugSerial.printf("[debug] FPM: Error in receiving packet %u\n!", count);
            #endif
            return 0; 
        }
        yield();
    }
    uint16_t total_bytes = count * FPM::packet_lengths[params.packet_len];
    return total_bytes;
}
```

brianrho, 2020-12-06:
If you enable DEBUG prints in the midst of the template transfer itself, it can cause the process to fail as the sensor sends data faster than the MCU is processing it (because of your print statements). First try with the default templates example as is, and see if that works. If it doesn't, post the output from your *serial monitor*, not your sketch. 
But if it works, then try with your own sketch without debug enabled. Whatever the result, provide the serial logs.

closed 2023-12-26 by brianrho