brianrho/FPM #39: How can i get buffer code?

rickyanwar, 2021-01-09:
[image]
hi I am new in this topic , and also new about IoT, can you help can I get buffer code

brianrho, 2021-01-09:
What buffer code?

rickyanwar, 2021-01-09:
I mean buffer data
https://stackoverflow.com/questions/60443399/convert-buffer-data-to-an-image

, i need image of fingerprint, and send to backend, or you have any idea how to store template in database website and match form it, not save template in sensor

brianrho, 2021-01-09:
> Included is a Python 3 script for extracting fingerprint images to a PC; the image_to_pc example must first be uploaded to the Arduino. 

Did you follow these instructions in the README? If you did, then you should already have the fingerprint image on your PC. What you next do with it is up to you.

rickyanwar, 2021-01-09:
I was running but have probelm with port , status always busy, 

You have any idea how to save fingerprint template in cloud database and match in website, not matching from the sensor? Because fingerprint sensor have limit template it can be save, 🙏

brianrho, 2021-01-10:
> I was running but have probelm with port , status always busy,

I'm not sure what you're saying here, but make sure to close the Serial Monitor before you run the script, else the script will be unable to open the serial port to the sensor.

As for your other questions:

- You can save fingerprint templates to your PC, a database or wherever, by first extracting the templates from the sensor using the "templates" example. The example will print the byte contents of a template to the serial port, where you can view and save it as you like. You can obviously customize the example to other things with the template contents.
- In order to perform matching on the desktop, you would need either documentation on the algorithm/format used by the manufacturers for generating/matching templates *or* a ready-made library that implements these things already. At the moment, the former is unavailable, since it's probably a trade secret. What I've found so far is only (some of) the latter: a 32-bit Windows DLL (`ARTH_DLL.dll`) that exports only functions for matching two templates (`Match2Fp()`) and for generating templates from images (`GenChar()`). 

This DLL was found on some random site, and wasn't an official release from any manufacturer AFAIK; use at your own risk naturally. I don't know which exact sensor it was written for but I remember it at least seemed to worked well with my FPM10A sensor. Templates from other UART sensors like this may also work since they seem to all use the same API. Using this DLL, it should be possible for one to also match a single input user template against each template stored in some database i.e. 1-N matching.

[fpmatch](https://github.com/brianrho/fpmatch) contains an example on how to match two templates using a Python script that loads the DLL. There's also some C code in there that does the same thing. So if your server runs Windows, then you already have a decent starting point here.

However, if your server runs some Linux variant, then this solution cannot work as-is since DLLs are a Windows thing and cannot be used natively on Linux, especially if the DLL calls into Windows builtin libraries. So there's presently no Linux solution.

But if you're up for experimenting, you can check out loading the DLL under Wine/winelib if this is an option for your server. There's also another option: it seems someone's written a [library](https://github.com/taviso/loadlibrary) that enables Linux native code to load and call functions from Windows DLLs, *without needing Wine and similar*. If this is as good as advertised, then it may be possible for you to write C/C++ code on your Linux server, that loads the DLL and calls the right functions for template matching.

However, even this general solution using a shady DLL isn't the best, because all it takes for this solution to break is for your sensor manufacturer to decide to use new algorithms/formats for template generation or even change the existing one just enough to make it incompatible with this quite-old DLL. So if you really want to work safe, you may want to try at least contacting the sensor manufacturer and requesting an official library from them, which should come with guarantees that they'll keep it updated if anything changes sensor-side.

The discussions in this particular [thread](https://github.com/brianrho/FPM/issues/9#issuecomment-426583577) and the other [closed](https://github.com/brianrho/FPM/issues?page=1&q=is%3Aissue+is%3Aclosed) issues may also be interesting to you, though some of the info in there may be obsolete. Good luck.

rickyanwar, 2021-01-10:
Thank you for this information

closed 2021-01-10 by rickyanwar