brianrho/FPM #41: Upload prints to finger module

danbicks, 2022-01-10:
Hi Brian been a long time.
I am revisiting an old project and have the R503 reader types. Your library is excellent and all of my old code ported over fine.

Currently I have stored all the finger prints from the module to SD card in separate files, all looks good I can read these all back and the data is the same as downloaded. Total byte size is 1536.

I want to now load these back in to a blank R503 Reader, can you tell me the functions I need to call in order to achieve this function please.

This is an example of the first finger bytes loaded from the array. `Template 1 loaded
Printing template:
---------------------------------------------
2 5 99 5B 8B 2E A3 0 84 9B 9 9B 0 7E A 1F
55 45 3F C6 2A D6 F1 4F F1 F0 FD EC 53 FC 86 72
97 14 FF 27 47 E0 85 3F CC 47 B9 11 4F F3 63 A2
A0 53 FC DC BB 96 14 FF 39 94 2F 85 3F FF 8E 8B
10 5B 0 84 88 F 86 41 AF 5 9D D4 45 1A 4 8B
16 31 46 41 C4 A5 E0 52 2C 72 47 75 14 5F 20 20
A2 85 12 4A 2 4B 41 46 A2 B8 5E 90 52 24 B0 8F
83 14 4D 2D 3E A6 5 1A 8B ED 18 C1 45 93 50 7A
EC 51 B8 DA 68 B2 14 50 3F C 68 85 1A D4 AF B9
B1 48 85 A0 63 20 52 3F 8E 8B B0 C3 0 84 DA 16 
D9 F6 AE 10 4F C 85 1B 86 B4 11 E1 45 F1 C3 13
DC 51 54 78 4D CF 14 6F 27 A2 38 45 13 8A 72 51
B1 43 E2 A3 98 A4 51 8 D7 42 4C 14 3F 37 35 C2
48 4F CF 83 6B 51 46 4 23 F9 38 50 F9 12 E5 A
14 3E 45 D5 D0 85 13 51 8E A5 81 43 F4 64 24 F8
51 5D 1D 2E 5F 14 43 4D 22 28 C5 12 93 8A C3 31
44 D4 E8 A2 6C 51 A5 40 FD 48 14 3F 51 B8 9B 5
12 55 4A A8 41 44 9F 8E 8B B2 CB 0 84 DA 13 DA
20 B2 D BE C3 85 21 47 33 23 21 46 D2 6C CC 78
51 6C 9E B8 F5 14 50 28 90 75 C5 1C C A7 AD F1 
45 13 63 60 A4 51 C D9 19 1D 14 3D 41 CD 13 5
12 50 AC 90 91 B4 44 90 9A DC 51 A5 36 EC 52 14
54 4F B4 40 85 14 14 54 73 E1 45 75 23 DD 8 51
65 4C DA 58 14 5B 53 C7 97 C5 15 16 47 6A D1 46
75 AA 68 54 51 6F 8E 8B 73 E3 0 84 B7 10 B6 4A
B2 12 B5 44 5 21 8 50 C3 21 46 C2 BC 44 78 51
68 B2 94 F7 14 51 2D 87 75 C5 1E 8D E5 6E 11 46
3 AA F8 90 50 FC EA F3 1D 14 3D 47 44 93 5 12
D1 EA 50 91 44 45 33 31 40 51 CD 52 B1 6 14 70
56 48 D0 85 1B 15 AD 54 1 47 A5 7B F1 7C 51 A1
60 B8 42 14 75 8E 8B B6 DB 0 84 DB 14 DB C7 76 
E BF C3 85 21 47 93 83 41 46 D2 7C D4 88 51 74
A2 BC F7 14 50 29 90 F5 45 1B 8C E7 EE 31 44 F3
73 70 AC 51 8 DD 1B 1F 14 3C 42 CD 93 85 12 10
CC B0 51 D4 54 90 AA E4 51 8D 3C F0 50 14 55 50
B5 C1 85 15 94 AF 94 1 45 A5 2D 25 8 51 5D 4E
DC 58 14 5A 54 48 18 45 15 16 47 8B 11 46 25 A0
62 8C 51 DD 6C 9C 13 14 57 8E 8B B2 DB 0 84 D4
18 D5 17 B5 7 CD C7 5 18 2 B 8F 71 45 90 91
7 4C 51 CC 4E 7C E3 14 56 16 38 89 85 10 85 B1
91 D1 44 22 2C D1 38 51 10 8E C9 9 1F 43 29 A 
AD C5 18 8B CE E5 1 44 C3 B 44 18 51 2C C9 45
42 14 49 32 BD D0 5 14 4D D A5 81 44 83 44 71
7C 51 8 E6 76 A7 14 41 3A 6 27 45 18 8F 8A 67
1 44 64 1D C9 C0 51 9D 8 96 80 14 48 52 CB 1E
5 14 55 9 97 31 46 35 72 1D 58 51 99 62 A9 88
14 6A 8E A6 C1 47 45 5 C9 EC 51 81 44 FC C3 14
5D 51 AD C5 C5 16 95 71 EA 61 46 5F 8E 74 75 19
AC 51 ED 1E 44 84 14 61 4A 47 9B 5 1B 16 4D A7
81 46 85 99 1D BC 51 E7 8E 1D 5C C5 1B 3F 8E F3
55 80 51 AB 8E 8E 0 0 7B D 21 5 3A E7 7A 58
58 8A AB 22 3E 8E 40 8A 57 15 3D 7C 20 A C7 16 
40 86 CC A 4E 1D 3F 6E 4 8B B5 1D 3E 27 78 8A
97 14 3F 36 4 8B A2 14 40 4C 70 89 7B 24 3E 80
CC 9 CD 16 3F 70 0 A 52 1B 40 6B 8 A AB 1D
3E 74 34 A 4F 1E 4C 9E 74 A C8 22 3F 53 C4 8A
3B 20 42 6B 88 B B7 14 4D 60 CC B 41 14 57 4
84 8A 20 14 4C 19 8C 89 82 14 46 22 10 89 76 14
4A 7A A0 1 63 1D 3F 5D 90 81 41 14 41 63 60 1
6A 14 42 39 FC B AF 14 68 AE 60 9 65 14 45 C1
1C A 65 14 41 6E D8 88 E3 14 49 90 98 88 88 14
5F C4 20 89 66 14 57 A3 9C 8 7A 14 64 72 50 88 
B2 14 5C 3 6C 81 89 14 5C 24 4 0 5E 14 71 3
A4 1 78 14 5F D2 4C 8A 5C 14 4B EB 94 81 D9 14
65 4A 20 C B3 14 6C 1A 58 89 7C 14 4D 1 E4 1
7D 14 53 B7 28 C C3 14 57 D6 C8 A D3 14 4F 8B
0 8D BE 14 73 EB 14 A D9 14 5D F4 54 3 5D 14
66 5F 38 C B6 14 5D D6 E8 8B CD 14 66 B3 14 85
C5 14 80 1F 74 8B 33 14 50 49 FC 89 8B 24 3D 11
34 2 91 14 3F C3 10 8D C9 14 86 F7 5C C D4 14
88 BB E8 2 88 14 4F DD 8 8D CF 14 8F B2 1C 81
73 14 51 ED D8 9 E7 14 6B 4A 5C 8A 95 25 40 C
35 4 DD 14 8A 27 10 8B 9F 14 49 7E 10 8B 9F 14 
4B 3C 10 84 3D 14 69 7E FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
--------------------------------------------
1536 bytes read.`

Not sure if I can reduce this array size settings for this that work are:
#define BUFF_SZ    36864UL //768   // R305 BUFFER SIZE ELSE 512 STANDARD
uint8_t template_buffer[BUFF_SZ];

Cool project, really having some fun with this. Any pointers will be brilliant. Huge thanks Dans

danbicks, 2022-01-11:
Hi Brian, you can close this I have worked it out and have it operational :) Big thanks Dans

brianrho, 2022-01-11:
Hi Dan, glad to hear you've got it working. Just in case, `move_template()` in the `templates` example will probably help here if you need to confirm steps. Good luck :-)

closed 2022-01-11 by brianrho

danbicks, 2022-01-12:
move_template() has all the answers :) Ty

danbicks, 2022-01-12:
Out of curiosity 1536 bytes is the template payload it seems. Are there any pointers on how I can display to an actual tft the fingerprint images based on the array data. Would be a really nice addition. I am using the tft_eSPI library. Have you experimented with images recreation on tft displays before?

brianrho, 2022-01-15:
Yeah, I did this once, some years back. Once you have a way to read the image from the sensor (from the examples), you can potentially write said image to a display simultaneously. 

As usual, It will take a few seconds though, to read the image from the sensor since the UART transport isn't the fastest. Keep in mind, what you have there is the *template* size, not the image size which is usually much greater at ~37 kB for 256x288 "4-bit" grayscale, depending on your sensor. You can't get any sort of image from the template data alone, since that would require interpreting it -- completely infeasible without info from the manufacturer who won't be telling you anything. And I'm not sure even they can get a useful image from the template, being a condensed record of your fingerprint's relevant characteristics.

brianrho, 2023-12-26:
> Have you experimented with images recreation on tft displays before?

Just to update this issue -- finally added an example for this (`image_to_tft`).