brianrho/FPM #42: Can not connect AS608 to Arduino UNO

FaterYU, 2022-03-03:
Hello! Thanks for your useful code, but I seem meet some problems. I try to solve it by searching much website by failed. 
I use Arduino UNO to control the AS608, when I upload the enroll example code, I get some mistake:
```C
Adafruit Fingerprint sensor enrollment
Did not find fingerprint sensor :(
```
On the other hand, I try the same program on Arduino Nano. Its mistake is same to UNO's.

And I have check my line connection, it seems without problems:
- Green link to 2
- White link to 3
- VDD link to 3.3v
- GND link to GND 

I try to put ``` finger.begin(57600); ``` into loop function, but it is helpless.

My main program is the enroll code which in example folder

Hope I can get some useful suggestions to solve my problems, thanks!

brianrho, 2022-03-04:
The wire connections in the examples were written for R30x/FPMxx sensors -- they may not necessarily be the same for your AS608. I haven't used this sensor before but judging by the tutorials I found, it follows the same UART protocol as FPM.

Try out these tutorials: https://electropeak.com/learn/interfacing-fpm10a-as608-optical-fingerprint-reader-sensor-module-with-arduino/, https://makersportal.com/blog/2019/6/9/arduino-optical-fingerprint-sensor-as608

They use the Adafruit library and explain how to connect the sensor to your Arduino-- follow the exact steps and make sure to use 3.3 V for power as recommended. If these tutorials work for you, then you can switch to using the FPM library if you like.

FaterYU, 2022-03-15:
Thanks! I found the reason why it is wrong. The AS608 is locked, so it can not connect to computer without authority.

closed 2022-03-15 by FaterYU