brianrho/FPM #43: Unable to save image using c language

kumar-ashish1, 2022-03-30:
Hi i am getting some problem while trying to save image using c getting problem in writing header , color platte and image buffer to file in c lanuage. Pls help

brianrho, 2022-03-30:
I'm afraid I can't help you with your own code. You'll have to go study the existing Python script and then think/do some research on porting it to C. If you have any more _precise_ questions about the script itself, I may be able to answer.

kumar-ashish1, 2022-03-31:
Hi i have done lots of r&d and finally got success with a little bit problem.In python i am getting bright image as compare to clanguage  . I know its due to color palatte so can u help me pls.
[image]
[image]

kumar-ashish1, 2022-03-31:
Ok thanks . After doing lots of r&d finally i have completed my code. Thank-you so much .

brianrho, 2022-03-31:
Glad to hear it, good luck!

closed 2022-03-31 by brianrho