brianrho/FPM #46: Resulting Image

WanLoq, 2022-12-08:
Hi, i hope my question does not sound dumb but please can it be used to scan any image regardless of if it is a fingerprint or something like a single character or word on paper?

brianrho, 2022-12-10:
No, it is a good question 🙂️

First, I haven't deliberately tried scanning anything but fingers, so I can't offer any concrete answers.

Next, we can try to guess what would happen, based on known behaviour so far. I'll assume you're referring specifically to the class of optical sensors like the R307. And that you want to take photos of something and dump those images from the sensor buffer.

Whenever you scan a something with the sensor, it returns a confirmation code which can be any of these from the datasheet:

```
0x00: finger collection success
0x01: error when receiving package
0x02: can’t detect finger
0x03: fail to collect finger
```

The last 2 suggest that it can tell whether you have placed a _finger_ or not. But it could also be that the language isn't as precise as it appears, and they actually mean that they can tell when you have placed _anything_ on the sensor. In the latter case, it may then be possible to fool the sensor into thinking a finger is placed on the sensor, when we're actually taking scanning some paper or whatever. The sensor should return success here, if that works.

**Alternatively**, we simply ask the sensor to take a snapshot of whatever we put on it, and then **ignore the confirmation code entirely**. Next, we dump the image buffer to see what got captured. For this to work, we'd be assuming that the image buffer's contents are always valid after the GetImage command runs, even when the sensor didn't detect a finger (like error 0x02).

You could try these on a sensor and see what happens.

WanLoq, 2022-12-10:
Thanks a lot for your extensive answer i really appreciate. I will get one of the R307 to test this out for myself. I will be sure to update this once I've been able to test it out.

closed 2023-12-26 by brianrho