brianrho/FPM #47: image over udp python example?

NaMLiM, 2023-07-22:
hello, did u have an example python code for getting the fingerprint image over udp? i tried modifying your getImage.py code but the result is just a white image like this...

sorry i have never tinker with binary data so i don't know how u assemble it in the code

[image]

brianrho, 2023-07-25:
Hi, are you able to run the regular image-to-pc example at least?

What does your modified `getImage.py` look like? You can provide it through `pastebin.com`.

NaMLiM, 2023-07-25:
yes, i can send the image through serial using the example code.

this is the modified code https://pastebin.com/jyXfL08Z

this is wireshark showing my pc receive packet from udp port 9999
[image]

brianrho, 2023-07-26:
Okay, a few things:

* You need to re-write the socket-recv-to-file [like so](https://gist.github.com/brianrho/5fc931aab6bd211e60491944c8ec9593). 
* Since UDP has no connection setup or teardown, you'll need some way to tell the script when the image stream begins and when it ends (e.g. a `\t` marker like the original `getImage` script). But that's not too urgent.

Without such a marker, for now, you'll need to have the script running **first**, before you place your finger on the sensor to start the image-transfer. This should ensure that the UDP server receives every packet sent by the Arduino. 

When the Arduino stops sending data for 5s or the script has read enough data for an image, the script will end.

brianrho, 2023-07-26:
Keep in mind that Wireshark only shows the data being sent over the network, but not necessarily whether the script/application is reading/processing said data i.e. a closed port. There are no ACKs etc. with UDP -- best you can hope for when sending data to a closed port, is ICMP errors and I don't think these are guaranteed from all devices.

NaMLiM, 2023-07-26:
thank you ... it worked 👍

one more question... can you recommend me a fingerprint matching library for r307?
i tried using SourceAFIS(c#) but the similarity score is always bad
my sensor can match fingerprint just fine using the built in matcher.

this is the fp image taken from the sensor
[image]

brianrho, 2023-07-29:
You mean a library for PC-side matching? 

I briefly considered porting SourceAFIS a long time ago, but didn't get very far. Still, it should work so long as you feed it images correctly, though I haven't tried it myself.

You can also try [this](https://github.com/brianrho/fpmatch). It should enable you generate templates and perform 1:1 matching, using a DLL "offered" by the sensor manufacturers. Windows and Linux are supported.

brianrho, 2023-08-02:
I'll be closing this, if that answers your questions.

closed 2023-08-02 by brianrho