brianrho/FPM #50: Showing error : Connection Error

Repulsar19, 2023-11-22:
Hello, I'm using a HLK-FPM383F Fingerprint sensor for my project. I'm usning your code for enrolling and deleting the fingerprints images. All my connections are right and also fingerprint sensor is detected as well. But when I enter a slot number, it starts to show 'Communication Error' . What am I missing ?

brianrho, 2023-12-26:
Hi, if you still have the problem, can you try git-pulling and re-testing with the latest source? 

If the problem still happens:
-  change the value of `FPM_LOG_LEVEL` on this line to `FPM_LOG_LEVEL_V_VERBOSE`:
https://github.com/brianrho/FPM/blob/7ecf3489bbc56d8d22988100074eac60953983c8/src/fpm_logging.h#L24
- re-upload and redo the test with the Serial monitor open
- paste the entire contents of the Serial Monitor after the failure happens.

brianrho, 2024-02-02:
Closing this. You can re-open when you have the requested data, if you still have the problem.

closed 2024-02-02 by brianrho

Repulsar19, 2024-03-24:
Hello, thank you for your reply. Sorry I was not active so can't able to thank you earlier. Will test it and update you about it :)