brianrho/FPM #52: Error with R307, non-readable char on serial monitor

Ayush0Chaudhary, 2024-05-11:
In the serial monitor, it is showing these non-readable character.

[image]

## System
OS: Windows 10
Arduino Uno
Fingerprint sensor R306

brianrho, 2024-05-12:
You need to make sure you set the baud rate correctly in your serial monitor, to match the baud rate in the sketch -- 57600.

Ayush0Chaudhary, 2024-05-12:
It worked, you are OG, thanks!

Ayush0Chaudhary, 2024-05-13:
@brianrho hey, I wanted your help, I am a student at IIT Roorkee and working on an ambitious project for my college.

We want to create biometric system for 18000 people living in IIT Roorkee, we couldn't find any finger-machine that full fill our capacity, so we decided to make a machine that 
1. take bitmap image of our finger, 
2. sends it to server and then 
3. we run search for best match using knn and SHIFT
4. Return the result to our machine with yes or no.

**Do you think if this is a valid approach?**
**Does their exist easier solutions?**

I will be highly obliged if you answer this. :smiling_face_with_tear: :

brianrho, 2024-05-18:
Ah, that's an idea many have had.

You may want to first check out the old issues in this repo such as https://github.com/brianrho/FPM/issues/39 and https://github.com/brianrho/FPM/issues/9#issuecomment-451048453.

I don't know anything of "knn and shift", but if you replace that with "sensible algorithm that searches for prints in reasonable time", then yeah, your approach looks valid -- as in, it should work and yield the expected results.

How usable it will be in practice, is a separate question. With your proposed steps, you will want to think about:

* the time taken to extract the fingerprint image from the sensor (over UART, which is the common interface for such modules)
* the time taken to transfer that (e.g. 74 kB) image to your server for searching and matching
* the time taken on the server to search 18000 prints, which may not be a big deal comparatively, depending on your chosen algorithm and available compute

... all of which will need to happen before the user gets a yes/no.

I don't know of any easier paths, that are also sustainable. Perhaps you could try discussing with some sensor manufacturer to see if they're willing to supply you with their sensors and accompanying libraries (DLLs and such, such as [this one](https://github.com/brianrho/fpmatch)) for matching/searching prints. 

This would enable you perform the template generation in the sensor itself (using their builtin algo), offload the much-smaller template to your server, and use their libraries for the server-side matching. Of course, if your vendor ever goes belly-up, you're screwed. So maybe you'll want to store all the images on your server, as insurance, just in case you ever need to find a new vendor.

If all that's not safe enough for you, you may want to try developing your own matching algorithm(s) after all, then optimize the hell out of the path as you transfer the image from sensor to server for the search. Good luck.

closed 2024-05-25 by brianrho

Ayush0Chaudhary, 2024-05-26:
Thank you sir for replying, I will reply when we achieve a milestone on this project.