brianrho/FPM #53: udp server

ariessetiyawan, 2024-08-22:
how to create a udp server in python to receive packets and write back to bmp files

thanks

brianrho, 2024-08-26:
Your request doesn't make sense. Receive packets from whom? What kind of packets?

ariessetiyawan, 2024-08-26:
I'm trying to take an image from the fingerprint sensor and send it to the server/PC via UDP protocol, and I tried using the example script image_to_udp.ino, but the result is 32000 bytes stream not 36864

.
.
.
.
Image taken.
Starting image stream...
[+]Wrong checksum: 0xFFFF != 0x756F

32000 bytes transferred.

brianrho, 2024-08-26:
Okay, provide details of your setup. 

- What MCU/board(s) are you using? Is your UDP over WiFi or Ethernet? Are you using some special module?
- Download or clone the **latest** version of the library and make sure you're using that for your test.
- Make sure that you don't add any new `Serial.print()` calls inside the sketch or in the libraries.
- If the transfer is still failing, does it **always** fail after _exactly_ 32000 bytes? Or does the count change from test to test?
- Paste here the entire contents of your Serial Monitor after you've run the test. [Make sure to format your paste properly, using [backticks](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code-and-syntax-highlighting) -- do not just dump the text.]
- Also, paste the entire output of your UDP server on the PC, similarly formatted.

ariessetiyawan, 2024-08-28:
Thank you for your attention..my problem has been resolved.

I use an ESP32 board with a ZA650_M5 fingerprint sensor, network connection using WIFI

The following is the source code in the Arduino IDE and UDP server in Python

https://github.com/ariessetiyawan/send_fingerprint_image_UDP

brianrho, 2024-08-28:
Nice. What was the issue in the end?

closed 2024-08-28 by ariessetiyawan