brianrho/FPM #54: LED Control for ZW111 Fingerprint sensor

Repulsar19, 2024-09-13:
Hello, I have tried using your library and Im using ZW111 Fingerprint sensor. Though the code is working fine, i cant able to control LED for it. :(

brianrho, 2024-09-15:
Hi. As explained in `fpm.h`, only a few sensors are known to allow LED control:

```
    /* these 3 have been tested successfully only on ZFM60 so far. 
       May yet work on other/newer sensors */
    FPMStatus ledOn(void);
    FPMStatus ledOff(void);
```

I cannot find your sensor's datasheet, so unless you have some evidence that it supports such control as well as a description of the necessary commands, then it's best to assume it doesn't have that support.

closed 2024-09-19 by brianrho

Repulsar19, 2024-09-23:
hello sir sorry for late reply, i have got a datasheet of same fingerprint module (zw111), with commands list can you help me in this? should i send it to you here?

brianrho, 2024-10-01:
Sure, you can attach the PDF here.