brianrho/FPM #57: Upload template to the sensors flash

eeyosii, 2024-10-29:
i want to save the templates i got from the example "Templates" to the sensors flash, so then i can use it to delete the templates stored on the sensor and replacing them with new ones.

brianrho, 2024-10-30:
I don't understand -- you want to save templates into the sensor's flash (from where?) and then delete those same templates later?

eeyosii, 2024-10-30:
Thanks for your reply. 
[image]
My problem is that after enrolling fingers to the sensor using this library, I extracted the templates of enrolled fingers on the sensor, copied them, and saved them to a header file on my computer. I'm now attempting to figure out how to re-enroll the templates that were taken out of the sensor, instead of enrolling with finger again.

brianrho, 2024-10-30:
You can still use the same DLL, assuming your 768-byte template is composed of 3 256-byte feature-files.

As you can see in `fpmatch.py`, the matching function is called with each feature file hunk of 256 octets within the template:

```
        # match against each feature file within the template
        first = arth.Match2Fp(bytes(userInput), bytes(template))
        second = arth.Match2Fp(bytes(userInput), bytes(template)[FPM_CHAR_SIZE:])
```

So, all you have to do is add another line to match/test the 3rd feature file in your template:

```
        third = arth.Match2Fp(bytes(userInput), bytes(template)[FPM_CHAR_SIZE+FPM_CHAR_SIZE:])
```

... and take the average score of all 3 match calls.

Of course, you can also do all this in a loop.

(Also, don't forget to change `FPM_TEMPLATE_SIZE = 768` to be safe.)